ETH3D/badslam #10: Integrate surface reconstruction approach(es)

puzzlepaint, 2019-06-18:
It is often desirable to obtain a surface (mesh) reconstruction rather than only a point cloud, which is the current reconstruction output of BAD SLAM. Some possible options might be:

* Integrate [Poisson Surface Reconstruction](https://github.com/mkazhdan/PoissonRecon) as a post-processing step (such that it can be conveniently invoked with a single click from the GUI)

* Integrate [SurfelMeshing](https://github.com/puzzlepaint/surfelmeshing/) for live meshing during the reconstruction. A good integration might not be straightforward however, since:
  * BAD SLAM is constantly creating new surfels and merging surfels in existing areas, which SurfelMeshing might have trouble keeping up with.
  * For SLAM, one wants to sparsify the surfels a bit to keep it fast, while for reconstruction, the surfels should be as dense as possible.

kirbiyik, 2020-01-13:
Is there any update on this feature @puzzlepaint ?

puzzlepaint, 2020-01-13:
I created the issues with the "enhancement" label to show what would be useful additions in my opinion, but this does not mean that I have any plans to work on these. So, no updates should be expected. If anyone wants to contribute a PR with those features, it would however be welcome of course.

Regarding this particular issue, note that it is already possible to create a densified point cloud, save it to disk, and run something like Poisson Surface Reconstruction on it manually to get a mesh reconstruction.