ETH3D/badslam #2: Change the storage of keyframes from a std::vector to std::unordered_map

puzzlepaint, 2019-06-18:
Currently, the keyframes are stored in a std::vector, and removing keyframes leaves nullptr entries in that vector. This is likely not to be handled correctly in all parts of the codebase, leading to crashes. It would most likely be preferable to store the keyframes in an unordered_map instead (mapping keyframe_id -> keyframe_pointer) and delete removed keyframes from it entirely.

Young532042725, 2019-06-19:
@puzzlepaint hi, there are some problem occur, when building(make).  would  you  please give me some advice?  thank you very much!   There are cmake  and  make

 cmake -DCMAKE_CUDA_FLAGS="-arch=sm_61" ..
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   serialization
--   filesystem
--   system
-- Symbol prefix: 
-- Boost version: 1.58.0
DLib library found, using it from the system
CMake Warning at applications/badslam/CMakeLists.txt:21 (find_package):
  By not providing "Findrealsense2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "realsense2", but CMake did not find one.

  Could not find a package configuration file provided by "realsense2" with
  any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

CMake Warning at applications/badslam/CMakeLists.txt:24 (find_package):
  By not providing "Findk4a.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "k4a", but
  CMake did not find one.

  Could not find a package configuration file provided by "k4a" with any of
  the following names:

    k4aConfig.cmake
    k4a-config.cmake

  Add the installation prefix of "k4a" to CMAKE_PREFIX_PATH or set "k4a_DIR"
  to a directory containing one of the above files.  If "k4a" provides a
  separate development package or SDK, be sure it has been installed.

CMake Warning at applications/badslam/CMakeLists.txt:25 (find_package):
  By not providing "Findk4arecord.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "k4arecord", but CMake did not find one.

  Could not find a package configuration file provided by "k4arecord" with
  any of the following names:

    k4arecordConfig.cmake
    k4arecord-config.cmake

  Add the installation prefix of "k4arecord" to CMAKE_PREFIX_PATH or set
  "k4arecord_DIR" to a directory containing one of the above files.  If
  "k4arecord" provides a separate development package or SDK, be sure it has
  been installed.

-- The following OPTIONAL packages have been found:

 * PythonInterp

-- The following REQUIRED packages have been found:

 * ZLIB
 * GLEW
 * Qt5Core
 * Qt5Gui (required version >= 5.5.1)
 * Qt5OpenGL
 * Qt5Widgets
 * Qt5
 * Qt5X11Extras
 * SuiteSparse
 * Eigen3
 * g2o
 * OpenGL
 * Threads
 * X11
 * OpenCV
 * Boost
 * DLib
 * opengv

-- The following OPTIONAL packages have not been found:

 * realsense2
 * k4a
 * k4arecord

-- Configuring done
-- Generating done
-- Build files have been written to: /home/gp/BAD-SLAM/badslam/build
[image]

puzzlepaint, 2019-06-19:
Hi, this comment is unrelated to the original topic of this issue. I thus moved it to a new issue: #16