ETH3D/badslam #16: Build issues: std::atomic and Qt::AA_DontCheckOpenGLContextThreadAffinity not defined

puzzlepaint, 2019-06-19:
@puzzlepaint hi, there are some problem occur, when building(make).  would  you  please give me some advice?  thank you very much!   There are cmake  and  make

 cmake -DCMAKE_CUDA_FLAGS="-arch=sm_61" ..
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   serialization
--   filesystem
--   system
-- Symbol prefix: 
-- Boost version: 1.58.0
DLib library found, using it from the system
CMake Warning at applications/badslam/CMakeLists.txt:21 (find_package):
  By not providing "Findrealsense2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "realsense2", but CMake did not find one.

  Could not find a package configuration file provided by "realsense2" with
  any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

CMake Warning at applications/badslam/CMakeLists.txt:24 (find_package):
  By not providing "Findk4a.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "k4a", but
  CMake did not find one.

  Could not find a package configuration file provided by "k4a" with any of
  the following names:

    k4aConfig.cmake
    k4a-config.cmake

  Add the installation prefix of "k4a" to CMAKE_PREFIX_PATH or set "k4a_DIR"
  to a directory containing one of the above files.  If "k4a" provides a
  separate development package or SDK, be sure it has been installed.

CMake Warning at applications/badslam/CMakeLists.txt:25 (find_package):
  By not providing "Findk4arecord.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "k4arecord", but CMake did not find one.

  Could not find a package configuration file provided by "k4arecord" with
  any of the following names:

    k4arecordConfig.cmake
    k4arecord-config.cmake

  Add the installation prefix of "k4arecord" to CMAKE_PREFIX_PATH or set
  "k4arecord_DIR" to a directory containing one of the above files.  If
  "k4arecord" provides a separate development package or SDK, be sure it has
  been installed.

-- The following OPTIONAL packages have been found:

 * PythonInterp

-- The following REQUIRED packages have been found:

 * ZLIB
 * GLEW
 * Qt5Core
 * Qt5Gui (required version >= 5.5.1)
 * Qt5OpenGL
 * Qt5Widgets
 * Qt5
 * Qt5X11Extras
 * SuiteSparse
 * Eigen3
 * g2o
 * OpenGL
 * Threads
 * X11
 * OpenCV
 * Boost
 * DLib
 * opengv

-- The following OPTIONAL packages have not been found:

 * realsense2
 * k4a
 * k4arecord

-- Configuring done
-- Generating done
-- Build files have been written to: /home/gp/BAD-SLAM/badslam/build
[image]

_Originally posted by @Young532042725 in https://github.com/ETH3D/badslam/issues/2#issuecomment-503474158_

puzzlepaint, 2019-06-19:
Hi, the warnings in the CMake output can be ignored as long as you don't want to use RealSense or K4A live input. These libraries are optional.

The enum value Qt::AA_DontCheckOpenGLContextThreadAffinity has been added in Qt 5.8 according to the Qt documentation. It seems that you are likely using an older version? I will change the CMakeLists.txt file and ReadMe to require at least Qt 5.8.

To fix the error about std::atomic having an incomplete type, it should be sufficient to add `#include <atomic>` at the top of the file qt_thread.cc in which the error occurs. I will also fix that in the code.

Young532042725, 2019-06-20:
@puzzlepaint  Hi， Thank U for your reply! i have installed Qt 5.8.0 using (qt-opensource-linux-x64-5.8.0.run) from Official website.  and also modified the CMakeLists.txt as "find_package(Qt5 5.8 REQUIRED COMPONENTS Core OpenGL Widgets)", but the same problem still occurs!! as follow: (May be my  Qt 5.8.0 is not OK, can you give me some instruction for installing it?)

/home/gp/BAD-SLAM/badslam/libvis/src/libvis/opengl_context_qt.cc:126:46: error: ‘AA_DontCheckOpenGLContextThreadAffinity’ is not a member of ‘Qt’
   QCoreApplication::instance()->setAttribute(Qt::AA_DontCheckOpenGLContextThreadAffinity);

Thank YOU very much!

Young532042725, 2019-06-20:
@puzzlepaint  it find Qt5.8 by "SET(Qt5_DIR "/home/gp/Qt5.8.0/5.8/gcc_64/lib/cmake/Qt5/")
SET(Qt5Test_DIR "/home/gp/Qt5.8.0/5.8/gcc_64/lib/cmake/Qt5Test")" in CMakeLists.  but another problem occurs!! 
[image]

Young532042725, 2019-06-20:
@puzzlepaint, Dear friend, would you please write down the commands for the whole building process in sequence in Readme?  Thank You very much!

puzzlepaint, 2019-06-20:
It is also possible to set the Qt5_DIR and related variables via the CMake GUI or via ccmake. This avoids hardcoding them into CMakeLists.txt.

The ReadMe already contains all the build commands for the application. The errors that occur on your system arise from differing versions of the dependencies.

Regarding the new error (Eigen::half not found), it seems that fixing this requires to update the Eigen library. As stated in the ReadMe, the latest version (3.3.7) should definitely work, but I don't know what the minimum required version is.

Young532042725, 2019-06-21:
@puzzlepaint My Dear friend, i have solved the problem, but there still be a bug in mine. as follow. would you please provide some advice to me? Thanks for your help so much! Additional, really want to know how do you install the Qt 5.8? also from the *.run file? (if not, could you please give me some Specific Instructions or commands step by step. so unfamiliar to Qt). thank you again!!!
[image]

puzzlepaint, 2019-06-21:
This now looks like it might be an issue due to a too old version of the CUDA SDK. Which one do you use? Does it contain the __ushort_as_half and __half_as_ushort functions? All of the versions listed in the ReadMe should contain those.

I don't remember how I installed the up-to-date Qt, but I guess it probably was the .run file as well, yes.

puzzlepaint, 2019-07-16:
Closing this for now due to no further activity.

closed 2019-07-16 by puzzlepaint