ETH3D/badslam #18: make: *** No rule to make target 'badslam'.  Stop.

HuangTY96, 2019-07-22:
hi, when i execute "make -j badslam", why it said "make: *** No rule to make target 'badslam'.  Stop. "

HuangTY96, 2019-07-22:
i use ubuntu16.04, and I have installed all the dependency. After that I can run cmake successfully, and run make -j (without badslam) successfully, but I cannot get executable file.

puzzlepaint, 2019-07-22:
Hi, do you have the CUDA SDK installed (and does the CMake output indicate that it is found by CMake)? The `badslam` target is only added if `CMAKE_CUDA_COMPILER` is set.

HuangTY96, 2019-07-23:
I build the samples file in /usr/local/cuda-9.1 and delete the file build_RelWithDebInfo, then rebuild all the project, finally it works.

HuangTY96, 2019-07-23:
well, when I run ./badslam and choose dataset, then click start, I get this error:
./badslam: symbol lookup error: ./badslam: undefined symbol: _ZN5DBoW26FBrief10fromStringERN5boost14dynamic_bitsetImSaImEEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
could you please tell me what it is and how to solve this

HuangTY96, 2019-07-23:
My dear friend, now when I click start, It can work, but the result is totally horrible, and it gives me an error:
15:01:34.848        loop_detector.cc:355   INFO| - loop closure 3D-3D inlier count: 28
15:01:34.870        loop_detector.cc:601   INFO| - rotational_distance (0, 1): 0.000690534
15:01:34.870        loop_detector.cc:602   INFO| - translational_distance (0, 1): 0.000822317
15:01:34.870        loop_detector.cc:601   INFO| - rotational_distance (0, 2): 0.00231612
15:01:34.870        loop_detector.cc:602   INFO| - translational_distance (0, 2): 0.0025318
15:01:34.870        loop_detector.cc:601   INFO| - rotational_distance (1, 2): 0.00284715
15:01:34.870        loop_detector.cc:602   INFO| - translational_distance (1, 2): 0.00299204
15:01:34.870        loop_detector.cc:668   INFO| - average_pixel_distance: -nan
15:01:34.870 pose_graph_optimizer.cc:127   INFO| - Performing pose graph optimization ...
int g2o::csparse_extension::cs_cholsolsymb(const cs_di*, number_t*, const cs_dis*, number_t*, int*): cholesky failed!
Cholesky failure, writing debug.txt (Hessian loadable by Octave)
15:01:34.918 pose_graph_optimizer.cc:131   INFO| - Pose graph optimization done
Sophus ensure failed in function 'void Sophus::SO3Base<Derived>::normalize() [with Derived = Sophus::SO3<float>]', file '/home/hty/workspace/badslam/libvis/third_party/sophus/sophus/so3.hpp', line 166.
Quaternion should not be close to zero!
Aborted (core dumped)

this is the same error when I run the auto_tuning script, could you please tell me how to solve this, sorry to bother you, my friend :)

puzzlepaint, 2019-07-23:
As also commented in the other thread, please try setting the `BUILD_WITH_MARCH_NATIVE` CMake option to ON in your build of g2o.

HuangTY96, 2019-07-24:
Hi, my friend,
thanks for your help, I can run badslam now, but not all dataset on your website, for example, if I run dataset desk_2 and mannequin_4, it will give me totally wrong trajetory, and a lot of warning, If I select "enable loop detection", it will give me an error.(it is still this error: 
**int g2o::csparse_extension::cs_cholsolsymb(const cs_di*, number_t*, const cs_dis*, number_t*, int*): cholesky failed!
Cholesky failure, writing debug.txt (Hessian loadable by Octave)**)
I'm confused, is it about my GPU? My environment is Ubuntu16.04 with GTX1060 and CUDA9.1
[image]

puzzlepaint, 2019-07-24:
See https://www.eth3d.net/slam_benchmark : both for the desk_2 and mannequin_4 datasets, none of the methods seemed to give a good result, so failure on these datasets is expected.

HuangTY96, 2019-07-24:
> See https://www.eth3d.net/slam_benchmark : both for the desk_2 and mannequin_4 datasets, none of the methods seemed to give a good result, so failure on these datasets is expected.

Ok, thanks. So could you tell me the reason they fail. Now I want to record some data using my own RGB-D camera, and run these data on badslam, I can make the form of my data just like what we use on badslam, but I don't know how to make sure it won't fail, so could you give me some advice about this?

puzzlepaint, 2019-07-24:
Some of the most common failure reasons are:

* Camera views that don't fully constrain the camera pose (or with too little information for badslam to use it successfully), for example planar walls without much texture
* Too sparse depth images given by the depth camera
* Too fast camera movement (can be improved to some extent by increasing the keyframe frequency)
* Moving objects in the camera view (badslam assumes a static scene)

Young532042725, 2020-02-19:
@HuangTY96 hi, I have installed all the dependency,  cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_CUDA_FLAGS="-arch=sm_61" ..   
 it shows "CMake did not find CUDA, which is required to build this program.
" in terminal.  have u encounter this error??  thank you