ETH3D/badslam #19: Reconstruction configuration for custom dataset

jyiyang, 2019-07-22:
Thanks for this great project! I have no problem running with the eth3D dataset and get results. However, when I was trying another [dataset (Stanford copyroom)](https://drive.google.com/drive/folders/0B6qjzcYetERgSkJ3TFhVSDVDcU0), used by [Open3D](http://www.open3d.org/), I am having some troubles to get the results. 

Here is the structure of my files:
```
\copyroom
---\rgb
------000001.png
        ....
---\depth
------000001.png
        ....
---associated.txt
---calibration.txt
```

The format of depth image is 16 bit unsigned char, and the format of the color image is 8 bit unsigned char 3-channel. Since the unit of depth is in millimeter, I used the raw to metric depth scale of 0.001. The timestamp for the images are simply ordered as 1, 2, 3, 4...
I have tried disabling and enabling the motion model choice, but none of the options work. 

Here is a screen shot of the reconstruction interface. Would appreciated any help or advice on what configuration I should use instead. Thanks!

[image]

jyiyang, 2019-07-23:
Just found out that this is due to incorrect depth scale. After passing depth_scaling parameter the reconstruction looks fine. Sorry for the trouble.

closed 2019-07-23 by jyiyang

puzzlepaint, 2019-07-23:
Hi, glad that it works now. I am curious about the error with "u16 is not supported by the Qt image IO.", is that resolved now as well? It seems that the wrong image I/O implementation might have been used by the application there. It should use libpng for PNG files instead of Qt.

jyiyang, 2019-07-23:
After playing around with some of the data I found that the data I used reversed the order of color file and depth file in associated.txt...this seems to throw an error because the system is trying to read depth images as color images. The problem is resolved after I fix the order. Again, thanks for the great system!