ETH3D/badslam #21: Compiler errors

nonlinear1, 2019-08-06:
When I compile this slam, I was confronted with the following errors:
[ 94%] Building CXX object CMakeFiles/libvis_test.dir/libvis/src/libvis/test/lm_optimizer.cc.o
In file included from /home/manifold/badslam/./libvis/src/libvis/lm_optimizer.h:35:0,
                 from /home/manifold/badslam/libvis/src/libvis/test/lm_optimizer.cc:34:
/usr/include/cublasXt.h:57:10: fatal error: driver_types.h: no this file or directory
 #include "driver_types.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/libvis_test.dir/build.make:182: recipe for target 'CMakeFiles/libvis_test.dir/libvis/src/libvis/test/lm_optimizer.cc.o' failed
make[2]: *** [CMakeFiles/libvis_test.dir/libvis/src/libvis/test/lm_optimizer.cc.o] Error 1
Could anyone know why?
Thank you very much

nonlinear1, 2019-08-06:
I have solve this problem. change cmake to the latest version 3.15.1. everything is OK

closed 2019-08-06 by puzzlepaint

HTLife, 2019-09-18:
Even if I upgrade the CMake to CMake version 3.15.3,  the error still exists.

HTLife, 2019-09-18:
Need to add this line to CMakeLists.txt
```cmake
include_directories("/usr/local/cuda/include")
```

nonlinear1, 2019-10-06:
Have you solve your problem?

HTLife, 2019-10-06:
@nonlinear1 I still have no luck to solve this problem.

HTLife, 2019-10-06:
This is my environment setting (docker)
https://github.com/HTLife/badslam_env

nonlinear1, 2019-10-08:
Do you have wechat (微信)？　you can add. my wechat number is 31781410, then we can discuss your problem.