ETH3D/badslam #22: Pose estimation not converged ?

nonlinear1, 2019-08-06:
When I use Kinect for Azure run badslam, I got the following message:
6:03:53.713 direct_ba_alternating.c:249   WARN| Pose estimation not converged (not_converged_count: 253, call_counter: 35820)
16:03:53.756 direct_ba_alternating.c:249   WARN| Pose estimation not converged (not_converged_count: 254, call_counter: 36157)
16:03:54.082 direct_ba_alternating.c:249   WARN| Pose estimation not converged (not_converged_count: 255, call_counter: 36821)

could you anyone know why?

puzzlepaint, 2019-08-06:
The "Pose estimation not converged" warnings may sometimes be caused by oscillating pose estimation (which is not a big problem), if the scene doesn't sufficiently constrain a camera pose, or if too little depth data is available in a frame. Due to this, they sometimes also happen during normal operation and don't necessarily mean that something broke (unless such warnings are output permanently).

In your case, according to the messages, there were 255 warnings among 36821 calls to the pose estimation routine, which seems normal.

closed 2019-08-06 by puzzlepaint

nonlinear1, 2019-08-07:
Thanks a lot.