ETH3D/badslam #23: badslam_test failed with "Cuda Error: invalid device function"

barelycompiling, 2019-08-06:
Hello guys, good job! I faced the following problem:

When I launch ./applications/badslam/badslam_test, I get this message:

[==========] Running 14 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 14 tests from Optimization
[ RUN      ] Optimization.AlternatingGeometryOptimizationWithGeometricResidual
Stack trace:
15      0x7f98ad9fa41d clone + 109
14      0x7f98ae9716ba /lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba) [0x7f98ae9716ba]
13      0x7f98ae294c80 /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0xb8c80) [0x7f98ae294c80]
12      0x7f98aff42dae /home/me/Documents/bosch_projects/BAD_SLAM/BAD_SLAM_full/build/liblibvis.so(+0x86dae) [0x7f98aff42dae]
11            0x428e8f ./applications/badslam/badslam_test() [0x428e8f]
10      0x7f98b5bdafa0 testing::UnitTest::Run() + 64
9       0x7f98b5be2b43 bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) + 67
8       0x7f98b5bdac7f testing::internal::UnitTestImpl::RunAllTests() + 575
7       0x7f98b5bda9b5 testing::TestCase::Run() + 181
6       0x7f98b5bda8d8 testing::TestInfo::Run() + 280
5       0x7f98b5bda78a testing::Test::Run() + 186
4       0x7f98b5be2663 void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) + 67
3             0x4297b2 ./applications/badslam/badslam_test() [0x4297b2]
2             0x5650c5 ./applications/badslam/badslam_test() [0x5650c5]
1             0x44e9e1 ./applications/badslam/badslam_test() [0x44e9e1]
0       0x7f98afc9ed73 loguru::StreamLogger::~StreamLogger() + 163
11:15:59.004 cuda_image_processing.c:192   FATL| Cuda Error: invalid device function
Aborted (core dumped)

It is probably related to the fact that I use an Nvidia GeForce 950M graphic card with 5.0 compute capability. I also use CUDA 8.0. What can be done to get through this issue?

puzzlepaint, 2019-08-06:
Did you specify the correct CUDA code generation options, I guess probably -DCMAKE_CUDA_FLAGS="-arch=sm_50" for your graphics card?

Aside from that, as the ReadMe states, the program currently requires at least compute capability 5.3, but I believe that it is easy to change that (unless there are other things requiring a high compute capability that I am not aware of). The reason for requiring it is that point radii are stored as 'half' numbers in a memory buffer to save space, and the corresponding conversion functions require 5.3 according to [this table](https://docs.nvidia.com/cuda/cuda-c-programming-guide/index.html#features-and-technical-specifications). The alternative would be to store them as floats, which will have no negative consequences except requiring more GPU memory. To change this, one would need to look for all occurrences of the `radius_buffer` which is stored in the keyframes (keyframe.h header) and change its type from u16 to float. Then drop all of the conversions that happen when reading from or writing to this buffer. For example, in [cuda_depth_processing.cu#L355](https://github.com/ETH3D/badslam/blob/master/applications/badslam/src/badslam/cuda_depth_processing.cu#L355), simply drop the `__half_as_ushort(__float2half_rn())` around the value.

Young532042725, 2020-02-20:
@barelycompiling hi friends，i am now facing a similar error like yours.  I use an Nvidia GeForce GT730 graphic card. CUDA 10.0. Would you please give me some advice? Thanks!!
[image]

AreteQin, 2020-04-03:
Hi, there.

I am facing the same problem with you while doing auto_tunning. 

But I can't understand which file I should revise, could you please give me some help.

Thanks in advance!