ETH3D/badslam #25: BadSlam with Kinect Sensor V2 ?

Grungeby52, 2019-08-23:
Hi guys,

I have a Kinect Sensor v2 and I can read RGB-D which from sensor coming data with OPENNI library. Is that enough for use in BadSlam?

puzzlepaint, 2019-08-23:
There is no integration of OpenNI for live use in BAD SLAM (yet?), but in principle it should certainly work. You might look at the three existing integrated input libraries as examples for how to add an OpenNI integration:

* https://github.com/ETH3D/badslam/blob/master/applications/badslam/src/badslam/input_realsense.cc / h
* https://github.com/ETH3D/badslam/blob/master/applications/badslam/src/badslam/input_structure.cc / h
* https://github.com/ETH3D/badslam/blob/master/applications/badslam/src/badslam/input_azurekinect.cc / h

Alternatively, if offline processing of pre-recorded datasets is sufficient, you may convert the RGB-D data to the extended TUM RGB-D format (which is described in the ReadMe), which should work directly. Based on short googling, [this](https://github.com/chaowang15/RGBDCapture) might be an existing program to record TUM RGB-D data with OpenNI, although it might be missing determining of camera intrinsics and undistortion of the frames to pinhole.

Grungeby52, 2019-08-24:
I think I won't have any problems when I add the OpenNi library to the project and then adapt the required data structure. Thanks for answer...