ETH3D/badslam #27: Runtime error:

oafolabi, 2019-09-26:
Hi, 

I'm trying to run BADSLAM on one of the mannequin datasets and run into the following error: 
Do you have any idea what I can do ? Any help is appreciated, thanks.

Stack trace:
31      0x559502adb54a ./applications/badslam/badslam(+0x3054a) [0x559502adb54a]
30      0x7f4ee0b96b97 __libc_start_main + 231
29      0x559502ad3ed1 ./applications/badslam/badslam(+0x28ed1) [0x559502ad3ed1]
28      0x559502add471 ./applications/badslam/badslam(+0x32471) [0x559502add471]
27      0x559502b30052 ./applications/badslam/badslam(+0x85052) [0x559502b30052]
26      0x7f4ee2dffe95 QWidget::setVisible(bool) + 405
25      0x7f4ee2dfcd5f QWidgetPrivate::show_helper() + 63
24      0x7f4ee2dfccf0 QWidgetPrivate::showChildren(bool) + 240
23      0x7f4ee2dffe95 QWidget::setVisible(bool) + 405
22      0x7f4ee2dfcd5f QWidgetPrivate::show_helper() + 63
21      0x7f4ee2dfccf0 QWidgetPrivate::showChildren(bool) + 240
20      0x7f4ee2dffe95 QWidget::setVisible(bool) + 405
19      0x7f4ee2dfcd5f QWidgetPrivate::show_helper() + 63
18      0x7f4ee2dfccf0 QWidgetPrivate::showChildren(bool) + 240
17      0x7f4ee2dffe95 QWidget::setVisible(bool) + 405
16      0x7f4ee2dfcd43 QWidgetPrivate::show_helper() + 35
15      0x7f4ee2df91cd QWidgetPrivate::sendPendingMoveAndResizeEvents(bool, bool) + 445
14      0x7f4ee1f1ec18 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 264
13      0x7f4ee2dc99f0 QApplication::notify(QObject*, QEvent*) + 688
12      0x7f4ee2dc285c QApplicationPrivate::notify_helper(QObject*, QEvent*) + 156
11      0x7f4ee2e009c2 QWidget::event(QEvent*) + 2162
10      0x7f4ee2e1efc1 QOpenGLWidget::resizeEvent(QResizeEvent*) + 177
9       0x7f4ee2df9215 QWidgetPrivate::sendPaintEvent(QRegion const&) + 37
8       0x7f4ee1f1ec18 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 264
7       0x7f4ee2dc99f0 QApplication::notify(QObject*, QEvent*) + 688
6       0x7f4ee2dc285c QApplicationPrivate::notify_helper(QObject*, QEvent*) + 156
5       0x7f4ee2e00348 QWidget::event(QEvent*) + 504
4       0x7f4ee2e1f145 /home/towerman/Qt/5.12.0/gcc_64/lib/libQt5Widgets.so.5(+0x1bc145) [0x7f4ee2e1f145]
3       0x7f4ee3984ef9 vis::RenderWidgetOpenGL::paintGL() + 889
2       0x559502ca0313 ./applications/badslam/badslam(+0x1f5313) [0x559502ca0313]
1       0x559502c9cfae ./applications/badslam/badslam(+0x1f1fae) [0x559502c9cfae]
0       0x7f4ee36e4163 loguru::StreamLogger::~StreamLogger() + 163
14:39:46.997        render_window.cc:963   FATL| Cuda Error: unknown error

ishanivyas, 2019-11-02:
Hi I am having this error while running the dataset:

gui_main_window.cc:1600  INFO| Read dataset with 643 frames
QEGLPlatformContext: Failed to create context: 3009
QOpenGLWidget: Failed to create context
QEGLPlatformContext: Failed to create context: 3009
QOpenGLWidget: Failed to create context
QEGLPlatformContext: Failed to create context: 3009
QOpenGLWidget: Failed to create context

puzzlepaint, 2019-11-04:
Googling the error message `QEGLPlatformContext: Failed to create context: 3009` suggests that this might be due to old graphics hardware.

ishanivyas, 2019-11-06:
Ok I will look into it. Thanks.

ishanivyas, 2019-11-07:
@puzzlepaint  Now I am having this issue:

could be.
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
07:33:34.791      gui_main_window.cc:1600  INFO| Read dataset with 643 frames
Unrecognized OpenGL version
Unrecognized OpenGL version
[xcb] Too much data requested from _XRead
[xcb] This is most likely caused by a broken X extension library
[xcb] Aborting, sorry about that.
badslam: ../../src/xcb_io.c:741: _XRead: Assertion `!xcb_xlib_too_much_data_requested' failed.
Aborted (core dumped)

puzzlepaint, 2019-11-07:
This seems wrong: `libGL error: failed to load driver: swrast`, since that sounds like it is trying to load a software rasterizer. Which graphics card does your PC have? Is the driver for it installed properly?

ishanivyas, 2019-11-07:
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 430.26       Driver Version: 430.26       CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 108...  Off  | 00000000:03:00.0  On |                  N/A |
| 20%   36C    P8    10W / 250W |    237MiB / 11177MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   1  Quadro K5000        Off  | 00000000:82:00.0 Off |                  Off |
| 30%   37C    P8    14W / 137W |      1MiB /  4037MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
+-----------------------------------------------------------------------------+

puzzlepaint, 2019-11-07:
This looks like normal output to me rather than an error (apart from the fact that there seems to be no application listed with any GPU usage - for me, this lists for example /usr/bin/X). Thus, in case other OpenGL applications are working fine with that setup in the container, then I probably can't help. I have never tried to run the application with more than one GPU installed, no idea whether it handles that case properly.

ishanivyas, 2019-11-09:
I have removed older GPU and kept it NVIDIA-SMI 430.26 Driver Version: 430.26 CUDA Version: 10.2.  
I am not able to run any OpenGL application on my container. I am working on it to putting debugger in code. Let you know if I make any progress. Thank you for your time and help!

ishanivyas, 2019-11-09:
@puzzlepaint , After putting debugger, I got more details error message:
GNU gdb (Ubuntu 8.1-0ubuntu3.1) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./build_RelWithDebInfo/applications/badslam/badslam...done.
(gdb) run
Starting program: /home/ubuntu/badslam/build_RelWithDebInfo/applications/badslam/badslam 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
time                            file:line     v| 
23:39:08.780              loguru.cpp:587   INFO| arguments: /home/ubuntu/badslam/build_RelWithDebInfo/applications/badslam/badslam
23:39:08.780              loguru.cpp:590   INFO| Current dir: /home/ubuntu/badslam
23:39:08.780              loguru.cpp:592   INFO| stderr verbosity: 2
23:39:08.780              loguru.cpp:593   INFO| -----------------------------------
23:39:08.781                 main.cc:439   WARN| No auto-tuning file found -> using default parameters. GPU performance is thus probably slightly worse than it could b
[New Thread 0x7fffe4786700 (LWP 7206)]
[New Thread 0x7fffe28be700 (LWP 7207)]
[New Thread 0x7fffdab95700 (LWP 7208)]
[New Thread 0x7fffda394700 (LWP 7209)]
[New Thread 0x7fffd9b93700 (LWP 7210)]
[New Thread 0x7fffb9ffb700 (LWP 7217)]
[New Thread 0x7fffb97fa700 (LWP 7218)]
[New Thread 0x7fffb8ff9700 (LWP 7219)]
[New Thread 0x7fff9bfff700 (LWP 7220)]
[New Thread 0x7fff9b7fe700 (LWP 7221)]
[New Thread 0x7fff9affd700 (LWP 7222)]
[New Thread 0x7fff9a7fc700 (LWP 7223)]
libGL error: No matching fbConfigs or visuals found
[Thread 0x7fff9a7fc700 (LWP 7223) exited]
[Thread 0x7fff9affd700 (LWP 7222) exited]
[Thread 0x7fff9b7fe700 (LWP 7221) exited]
[Thread 0x7fff9bfff700 (LWP 7220) exited]
[Thread 0x7fffb8ff9700 (LWP 7219) exited]
[Thread 0x7fffb97fa700 (LWP 7218) exited]
[Thread 0x7fffb9ffb700 (LWP 7217) exited]
[Thread 0x7fffba7fc700 (LWP 7216) exited]
[Thread 0x7fffbaffd700 (LWP 7215) exited]
[Thread 0x7fffbb7fe700 (LWP 7214) exited]
[Thread 0x7fffbbfff700 (LWP 7213) exited]
[Thread 0x7fffd8b91700 (LWP 7212) exited]
[Thread 0x7fffd9392700 (LWP 7211) exited]
[Thread 0x7fffd9b93700 (LWP 7210) exited]
[Thread 0x7fffda394700 (LWP 7209) exited]
[Thread 0x7fffdab95700 (LWP 7208) exited]
libGL error: failed to load driver: swrast
[New Thread 0x7fff9a7fc700 (LWP 7224)]
[Thread 0x7fff9a7fc700 (LWP 7224) exited]
[New Thread 0x7fff9a7fc700 (LWP 7225)]
23:39:31.826      gui_main_window.cc:1600  INFO| Read dataset with 643 frames
Unrecognized OpenGL version
Unrecognized OpenGL version
[xcb] Too much data requested from _XRead
[xcb] This is most likely caused by a broken X extension library
[xcb] Aborting, sorry about that.
badslam: ../../src/xcb_io.c:741: _XRead: Assertion `!xcb_xlib_too_much_data_requested' failed.

Thread 1 "badslam" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
[... 18 lines cut ...]
#14 0x00007ffff3611c72 in QOpenGLWidget::resizeEvent(QResizeEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007ffff35f1bf2 in QWidget::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff35b283c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007ffff35ba104 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff28349c8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#19 0x00007ffff35ea0fb in QWidgetPrivate::sendPendingMoveAndResizeEvents(bool, bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#20 0x00007ffff35eded3 in QWidgetPrivate::show_helper() () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007ffff35f0beb in QWidget::setVisible(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#22 0x00007ffff35ede40 in QWidgetPrivate::showChildren(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#23 0x00007ffff35edeef in QWidgetPrivate::show_helper() () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x00007ffff35f0beb in QWidget::setVisible(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007ffff35ede40 in QWidgetPrivate::showChildren(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#26 0x00007ffff35edeef in QWidgetPrivate::show_helper() () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#27 0x00007ffff35f0beb in QWidget::setVisible(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#28 0x00007ffff35ede40 in QWidgetPrivate::showChildren(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#29 0x00007ffff35edeef in QWidgetPrivate::show_helper() () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
[... cut ...]
    import_calibration_path="", depth_scaling=5000, splat_half_extent_in_pixels=3, show_current_frame_cloud=false, follow_camera=false, show_input_images=false, 
    window_width=1280, window_height=720) at /home/ubuntu/badslam/applications/badslam/src/badslam/gui_main_window.cc:113
#32 0x000055555557f446 in __libvis_main (argc=<optimized out>, argv=0x7fffffffe418) at /home/ubuntu/badslam/applications/badslam/src/badslam/main.cc:478
#33 0x0000555555576421 in main (argc=1, argv=0x7fffffffe418) at /home/ubuntu/badslam/applications/badslam/src/badslam/main.cc:67

Any insight on it?

puzzlepaint, 2019-11-10:
This seems to go wrong in some Qt-internal place. There is probably nothing that BAD SLAM can do about it. If you can't run any OpenGL application with your setup as you wrote in your previous post, then this setup is the problem.

ishanivyas, 2019-11-17:
@puzzlepaint  I fixed the issue with OpenGL on host system. I can run glxgear on container. But now I am getting the error "Segmentation fault (core dumped)". Is it build.sh file for badslam available for? Not sure why I am having still trouble.

puzzlepaint, 2019-11-17:
Good that OpenGL works now. Unfortunately, "Segmentation fault (core dumped)" is a very generic error message that does not give any useful hint as to why the program crashed. At a minimum, a gdb backtrace would be required to make some guess.

ishanivyas, 2019-11-18:
Thank you for your response! 
When I did a gdb backtrace :

(gdb) bt
#0  0x0000000000000000 in ?? ()
#1  0x00007ffff41adf3a in vis::ShaderProgramOpenGL::AttachShader (this=this@entry=0x555556bf3578, 
    source_code=source_code@entry=0x5555557de498 "#version 150\nuniform mat4 u_model_view_projection_matrix;\nin vec4 in_position;\nin vec3 in_color;\nout vec3 var1_color;\nvoid main() {\n  var1_color = in_color;\n  gl_Position = u_model_view_projection_mat"..., type=type@entry=vis::ShaderProgramOpenGL::ShaderType::kVertexShader)
    at /home/ubuntu/badslam/libvis/src/libvis/shader_program_opengl.cc:84
#2  0x000055555561feea in vis::BadSlamRenderWindow::CreateSplatProgram (this=this@entry=0x555556bf30a0)
    at /home/ubuntu/badslam/applications/badslam/src/badslam/render_window.cc:1347
#3  0x000055555562031d in vis::BadSlamRenderWindow::Initialize (this=0x555556bf30a0) at /home/ubuntu/badslam/applications/badslam/src/badslam/render_window.cc:142
#4  0x00007ffff419efb2 in vis::RenderWidgetOpenGL::initializeGL (this=0x555556a58f90) at /home/ubuntu/badslam/libvis/src/libvis/render_window_qt_opengl.cc:55
#5  0x00007ffff362e5fd in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#6  0x00007ffff362ec72 in QOpenGLWidget::resizeEvent(QResizeEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#7  0x00007ffff360ebf2 in QWidget::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007ffff35cf83c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00007ffff35d7104 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007ffff28519c8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#11 0x00007ffff36070fb in QWidgetPrivate::sendPendingMoveAndResizeEvents(bool, bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007ffff360aed3 in QWidgetPrivate::show_helper() () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x00007ffff360dbeb in QWidget::setVisible(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007ffff360ae40 in QWidgetPrivate::showChildren(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007ffff360aeef in QWidgetPrivate::show_helper() () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff360dbeb in QWidget::setVisible(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007ffff360ae40 in QWidgetPrivate::showChildren(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff360aeef in QWidgetPrivate::show_helper() () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#19 0x00007ffff360dbeb in QWidget::setVisible(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#20 0x00007ffff360ae40 in QWidgetPrivate::showChildren(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007ffff360aeef in QWidgetPrivate::show_helper() () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#22 0x00007ffff360dbeb in QWidget::setVisible(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#23 0x00005555555c3824 in vis::ShowMainWindow (qapp=..., start_paused=<optimized out>, config=..., 
    program_path="/home/ubuntu/badslam/build_RelWithDebInfo/applications/badslam/badslam", dataset_folder_path="/home/ubuntu/badslam/training/mannequin_1", 
    import_calibration_path="", depth_scaling=5000, splat_half_extent_in_pixels=3, show_current_frame_cloud=false, follow_camera=false, show_input_images=false, 
    window_width=1280, window_height=720) at /home/ubuntu/badslam/applications/badslam/src/badslam/gui_main_window.cc:113
#24 0x000055555557f446 in __libvis_main (argc=<optimized out>, argv=0x7fffffffe428) at /home/ubuntu/badslam/applications/badslam/src/badslam/main.cc:478
#25 0x0000555555576421 in main (argc=1, argv=0x7fffffffe428) at /home/ubuntu/badslam/applications/badslam/src/badslam/main.cc:67

puzzlepaint, 2019-11-18:
Thank you for the backtrace. The program crashes at `/home/ubuntu/badslam/libvis/src/libvis/shader_program_opengl.cc:84`, which is the last line in this block:

```c++
  CHECK(shader_name != nullptr) << "Unknown shader type.";
  
  *shader_name = glCreateShader(shader_enum);
```

Since the first line in this block ensures that shader_name is not `nullptr`, the only way this can go to null is if `glCreateShader` is null. `glCreateShader` is a macro that expands to `GLEW_GET_FUN(__glewCreateShader)`. I guess that if this returns null, GLEW is probably not initialized correctly.

This is weird, since in `BadSlamRenderWindow::Initialize()`, GLEW is initialized as follows:

```c++
  GLenum glew_init_result = glewInit();
  CHECK_EQ(static_cast<int>(glew_init_result), GLEW_OK);
```

So if there was an error there, the program should abort there. Maybe GLEW doesn't report the error there for some reason.

I think that this is as much as I can diagnose from here. For some reason, GLEW doesn't work correctly, but I don't know what the reason is.

ishanivyas, 2019-11-26:
Hi @puzzlepaint,

I am trying to use AppImage binary for windows. We have correct version cuda and driver install for windows. But when we run auto_tune_parameter.py file it throws us error saying "cuda 8 driver version is insufficient for cuda runtime version" 

Could you please direct us? Do you have build.sh file to run on ubuntu?

Regards,
Ishani

puzzlepaint, 2019-11-26:
* The AppImage binary is for Linux, not Windows.
* Googling the CUDA error message suggests that this may be due to not having a recent graphics driver version installed ( https://stackoverflow.com/questions/41409842 ).
* There is no automatic build script file.

ishanivyas, 2019-11-26:
@puzzlepaint  Thank you for your help! Sorry its my mistake in message. I have tried .exe file for windows 10 and AppImage binary file for Linux (18.04).
Windows 10 has latest driver.

BinuxLiu, 2021-04-23:
> Hi,
>
> I'm trying to run BADSLAM on one of the mannequin datasets and run into the following error:
> Do you have any idea what I can do ? Any help is appreciated, thanks.
>
> Stack trace:
> 31 [...]

I solved this problem by re-install the graphics driver.
I used to install the driver manually. 
Although I can use 'nvidia-smi', my Linux information shows llvm.
I uninstalled it and automatically installed the driver through Linux. This error is gone. 
I've experimented many times, and it's true!