ETH3D/badslam #29: Compiling for arm?

C-Aniruddh, 2019-10-11:
Can this project be compiled on the Jetson nano?

puzzlepaint, 2019-10-11:
I don't know, I haven't tried.

magnusbarata, 2020-03-05:
It seems like arm is not yet supported. I have installed all the dependencies, and had succeeded on generating cmake files, but gets the `c++: error: unrecognized command line option ‘-msse2’` error on compile. Then I tried to remove `-msse` option from CMakeLists.txt and recompile but still no luck. Here is part of the error message:
```
[  0%] Built target Dependencies
[  0%] Generating pnglibconf.c
[  0%] Building CXX object CMakeFiles/loguru.dir/libvis/third_party/loguru/loguru.cpp.o
[  0%] Generating pngprefix.h
[  0%] Generating scripts/pnglibconf.c
[  0%] Generating scripts/symbols.out
[  0%] Generating pnglibconf.out
[  1%] Generating scripts/symbols.chk
[  3%] Generating pnglibconf.h
[  4%] Generating scripts/prefix.out
[  4%] Generating scripts/sym.out
[  6%] Generating scripts/vers.out
[  7%] Generating scripts/intprefix.out
[  7%] Generating libpng.sym
[  9%] Generating libpng.vers
[  9%] Built target genfiles
[ 10%] Automatic MOC and RCC for target DBoW2
[ 10%] Built target DBoW2_autogen
Scanning dependencies of target png
[ 10%] Building C object libvis/third_party/libpng/CMakeFiles/png.dir/png.c.o
[ 12%] Building C object libvis/third_party/libpng/CMakeFiles/png.dir/pngerror.c.o
[ 12%] Building C object libvis/third_party/libpng/CMakeFiles/png.dir/pngget.c.o
[ 12%] Building C object libvis/third_party/libpng/CMakeFiles/png.dir/pngmem.c.o
[ 14%] Building C object libvis/third_party/libpng/CMakeFiles/png.dir/pngpread.c.o
[ 14%] Building C object libvis/third_party/libpng/CMakeFiles/png.dir/pngread.c.o
[ 15%] Building C object libvis/third_party/libpng/CMakeFiles/png.dir/pngrio.c.o
[ 15%] Building C object libvis/third_party/libpng/CMakeFiles/png.dir/pngrtran.c.o
[ 17%] Building C object libvis/third_party/libpng/CMakeFiles/png.dir/pngrutil.c.o
[ 17%] Building C object libvis/third_party/libpng/CMakeFiles/png.dir/pngset.c.o
[ 18%] Building C object libvis/third_party/libpng/CMakeFiles/png.dir/pngtrans.c.o
[ 18%] Building C object libvis/third_party/libpng/CMakeFiles/png.dir/pngwio.c.o
[ 20%] Building C object libvis/third_party/libpng/CMakeFiles/png.dir/pngwrite.c.o
[ 20%] Building C object libvis/third_party/libpng/CMakeFiles/png.dir/pngwtran.c.o
[ 20%] Building C object libvis/third_party/libpng/CMakeFiles/png.dir/pngwutil.c.o
[ 21%] Building C object libvis/third_party/libpng/CMakeFiles/png.dir/arm/arm_init.c.o
[ 21%] Building ASM object libvis/third_party/libpng/CMakeFiles/png.dir/arm/filter_neon.S.o
[ 23%] Building C object libvis/third_party/libpng/CMakeFiles/png.dir/arm/filter_neon_intrinsics.c.o
[ 25%] Linking CXX shared library libloguru.so
[ 25%] Building C object libvis/third_party/libpng/CMakeFiles/png.dir/arm/palette_neon_intrinsics.c.o
[ 25%] Built target loguru
Scanning dependencies of target DBoW2
[ 25%] Building CXX object applications/badslam/third_party/DBoW2/CMakeFiles/DBoW2.dir/src/BowVector.cpp.o
[ 26%] Linking C shared library libpng16.so
[ 29%] Built target png
[ 29%] Building CXX object applications/badslam/third_party/DBoW2/CMakeFiles/DBoW2.dir/src/FBrief.cpp.o
[ 31%] Automatic MOC and RCC for target libvis
[ 31%] Built target libvis_autogen
Scanning dependencies of target libvis
[ 31%] Building CXX object CMakeFiles/libvis.dir/libvis/src/libvis/command_line_parser.cc.o
[ 32%] Building CXX object applications/badslam/third_party/DBoW2/CMakeFiles/DBoW2.dir/src/FORB.cpp.o
[ 34%] Building CXX object CMakeFiles/libvis.dir/libvis/src/libvis/glew.cc.o
[ 34%] Building CXX object CMakeFiles/libvis.dir/libvis/src/libvis/image.cc.o
[ 34%] Building CXX object applications/badslam/third_party/DBoW2/CMakeFiles/DBoW2.dir/src/FeatureVector.cpp.o
[ 35%] Building CXX object applications/badslam/third_party/DBoW2/CMakeFiles/DBoW2.dir/src/QueryResults.cpp.o
[ 35%] Building CXX object applications/badslam/third_party/DBoW2/CMakeFiles/DBoW2.dir/src/ScoringObject.cpp.o
[ 37%] Building CXX object applications/badslam/third_party/DBoW2/CMakeFiles/DBoW2.dir/DBoW2_autogen/mocs_compilation.cpp.o
[ 37%] Linking CXX shared library libDBoW2.so
[ 37%] Built target DBoW2
[ 39%] Building CXX object CMakeFiles/libvis.dir/libvis/src/libvis/image_display.cc.o
[ 39%] Building CXX object CMakeFiles/libvis.dir/libvis/src/libvis/image_display_qt_widget.cc.o
[ 40%] Building CXX object CMakeFiles/libvis.dir/libvis/src/libvis/image_display_qt_window.cc.o
[ 40%] Building CXX object CMakeFiles/libvis.dir/libvis/src/libvis/image_io.cc.o
[ 42%] Building CXX object CMakeFiles/libvis.dir/libvis/src/libvis/image_io_libpng.cc.o
[ 42%] Building CXX object CMakeFiles/libvis.dir/libvis/src/libvis/image_io_netpbm.cc.o
[ 43%] Building CXX object CMakeFiles/libvis.dir/libvis/src/libvis/image_io_qt.cc.o
[ 43%] Building CXX object CMakeFiles/libvis.dir/libvis/src/libvis/libvis.cc.o
[ 43%] Building CXX object CMakeFiles/libvis.dir/libvis/src/libvis/opengl.cc.o
[ 45%] Building CXX object CMakeFiles/libvis.dir/libvis/src/libvis/opengl_context.cc.o
[... 2054 lines cut ...]
                                                                                                                                                  ^
In file included from /home/optimizer/badslam/./libvis/src/libvis/opengl.h:48:0,
                 from /home/optimizer/badslam/./libvis/src/libvis/render_window_qt_opengl.h:32,
                 from /home/optimizer/badslam/libvis/src/libvis/render_window.cc:33:
[... cut ...]
 GLEW_FUN_EXPORT PFNGLPROGRAMUNIFORMMATRIX3X4FVPROC __glewProgramUniformMatrix3x4fv;
                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/aarch64-linux-gnu/qt5/QtGui/qopengl.h:105:0,
                 from /usr/include/aarch64-linux-gnu/qt5/QtWidgets/qopenglwidget.h:49,
                 from /usr/include/aarch64-linux-gnu/qt5/QtWidgets/QOpenGLWidget:1,
                 from /home/optimizer/badslam/./libvis/src/libvis/render_window_qt_opengl.h:34,

puzzlepaint, 2020-03-05:
Seems that GLEW conflicts with Qt's OpenGL support. One way to address this might be to try to separate the includes, such that no file includes both `glew.h` and Qt's `opengl.h`. Another, possibly simpler, way might be to drop GLEW altogether and use Qt's OpenGL functions instead. See the corresponding Qt documentation for [QOpenGLWidget](https://doc.qt.io/qt-5/qopenglwidget.html#details). Basically, one could obtain an OpenGL function object at the start of functions dealing with OpenGL, like:
```
QOpenGLFunctions* f = QOpenGLContext::currentContext()->functions();
```
and then prefix all `gl` functions with `f->`.

donalj, 2020-03-16:
On ARM architectures, the Qt5 version you get with apt-get install qt5-default is built with GLES which is where the conflicts arise. You can build Qt5 from source using following the steps [here](https://doc.qt.io/qt-5/linux-building.html) and at Step 3 do  
```
./configure -opengl desktop 
```
If you do this I would recommend cross-compiling as it took 4-5 hours for Qt5 to build on my Nano.

Make sure to also modify CMakeLists.txt as like @magnusbarata said, the `-msse*` options are for x86/x86-64 and is not supported on ARM. You can either remove these lines from the CMakeLists.txt or wrap them in an 

```
  if(NOT CMAKE_SYSTEM_PROCESSOR MATCHES "aarch64")
    set(LIBVIS_WARNING_OPTIONS "${LIBVIS_WARNING_OPTIONS};$<$<COMPILE_LANGUAGE:CXX>:-msse2>")
    set(LIBVIS_WARNING_OPTIONS "${LIBVIS_WARNING_OPTIONS};$<$<COMPILE_LANGUAGE:CXX>:-msse3>")
  endif()  ```

magnusbarata, 2020-04-19:
Thank you for your reply @puzzlepaint and @donalj. Both answers provide solution for compiling with jetson nano.  In the end, it seems like the slam work is too heavy for jetson nano.

tkircher, 2020-04-24:
@donalj Did you get this working on the Jetson Nano?

donalj, 2020-05-22:
> @donalj Did you get this working on the Jetson Nano?
@tkircher sorry for the late response. I did, but like mentioned above, it was too heavy for the Nano. I did manage to get it to run with some of the example datasets but when using physical cameras (kinect v2 and Realsense d435i) it couldn't handle it.

tkircher, 2020-05-22:
@donalj I just got the Kinect Azure DK working in RTAB-Map, just a few minutes ago.

https://github.com/tkircher/rtabmap

Should be merged in main tomorrow. I'm curious to know how you got badslam working on the Nano. I got it compiled but it never seemed to do anything. I didn't have the energy to track down whatever issue I was having, but it does make amazingly great point clouds on my i7 workstation.

tkircher, 2020-05-22:
I should mention, I'm especially curious in trying one more time to get badslam working, because I'm getting an Xavier NX soonish which has better performance.

jcyhcs, 2021-12-17:
@tkircher  
hi,professor, have you success run badslam on jetson xavier nx? i have success run on jetson nano, but jetson nano too weak

mississippia, 2022-08-01:
I have succeed compiled badslam on a TX2 NX and tested it with a Kinect Azure DK on realtime.
The key is to compile Qt5 form source with desktop configuration of openGL.
It will run but won't last.
Since Tx2 has a 4GB GPU memory, and badslam takes 3.5GB+ in initialization, and 10MB+ per second during running, if GPU memory safety threshold is set to 250 MB, then badslam can run for less than 2 minutes.
More to it , RGBD input won't work, use a IR image stream input.
[image]
It's my office, pretty abstract...

puzzlepaint, 2022-08-01:
Regarding the initial GPU memory consumption, notice that this is largely due to the pre-allocation of ample memory for surfel storage. However, under normal conditions, the default number of surfels won't be reached. In the screenshot above, only a tiny fraction of the 25 million pre-allocated surfels are actually used, as seen from the corresponding display in the window's status bar. To reduce the initial memory consumption, the number of surfels pre-allocated can thus be reduced. This may be done with the `--max_surfel_count` [command-line parameter](https://github.com/ETH3D/badslam/blob/master/applications/badslam/doc/command_line.md).

mississippia, 2022-08-03:
[image]
@puzzlepaint 
I changed the max surfel count, it works.
Now I can run with RGB stream input.
Tx2 is a little bit slow, badslam runs with a significant delay.