ETH3D/badslam #31: Release with support Azure Kinect

kirbiyik, 2019-11-18:
Currently pre-built binary for Windows do not include Azure Kinect option. I would really appreciate if you can release a version which includes Azure Kinect input option since I'm having problem with compiling currently.

kirbiyik, 2019-11-25:
@puzzlepaint any update on this? It's also OK if you provide an AppImage for Linux. In both Windows and Linux I am stuck at some point with dependencies and cannot proceed.

puzzlepaint, 2019-11-25:
I agree that it would be nice to release a binary with Azure Kinect support. However, I think that the application would possibly need to be modified a little first. The license terms of the Azure Kinect SDK aren't fully clear to me, so I think that it would be safest to let the users install that SDK separately, and load the necessary libraries dynamically if they are found on the user's PC, as opposed to the current way of linking directly to the library, to avoid re-distributing the SDK binaries which is perhaps not allowed. It doesn't help either that I don't have an Azure Kinect for testing.

So, as long as nobody implements that, there won't be a binary with Azure Kinect support. But feel free to open another issue about the problems with dependencies that you are facing. Maybe I or someone else can help. Yes, setting up the dependencies on Windows can be a pain, but if you use Ubuntu, it shouldn't be that hard on Linux.

kirbiyik, 2019-12-02:
Thanks, I understand your concerns about the license.

closed 2019-12-02 by kirbiyik