ETH3D/badslam #32: Manually-specified variables were not used by the project:      CMAKE_CUDA_FLAGS

AbbyRosewood, 2019-11-25:
I've tried to set the CMAKE_CUDA_FLAGS using "-DCMAKE_CUDA_FLAGS="-arch=sm_61" or other versions and I've had the CUDA10.1 installed with Ubuntu 14.04, yet I still couldn't get my cuda properly set.

I've also tried adding 
set(CMAKE_CUDA_FLAGS "${CMAKE_CUDA_FLAGS} -gencode arch=compute_61,code=sm_61")
message("CMAKE_CUDA_FLAGS: ${CMAKE_CUDA_FLAGS}")

to my CMakelist and it shows:
CMAKE_CUDA_FLAGS: -arch=sm_61 -gencode arch=compute_61,code=sm_61

AbbyRosewood, 2019-11-25:
(base) Abby@Abby-Precision-Tower-3620:~/study/badslam-master/build_RelWithDebInfo$ cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_CUDA_FLAGS="-arch=sm_61" ..
-- Symbol prefix: 
-- The following OPTIONAL packages have been found:

 * PythonInterp

-- The following REQUIRED packages have been found:

 * OpenGL
 * Eigen3
 * Boost
 * Threads
 * ZLIB
 * GLEW
 * Qt5Core
 * Qt5Gui (required version >= 5.9.7)
 * Qt5OpenGL
 * Qt5Widgets
 * Qt5 (required version >= 5.8)
 * Qt5X11Extras
 * OpenCV

-- Configuring done
-- Generating done
-- Build files have been written to: /home/Abby/study/badslam-master/build_RelWithDebInfo
(base) Abby@Abby-Precision-Tower-3620:~/study/badslam-master/build_RelWithDebInfo$ make -j badslam
make: *** No rule to make target `badslam'.  Stop.

puzzlepaint, 2019-11-25:
This looks like CMake does not find CUDA at all on your machine, since the badslam target doesn't seem to be created. This is probably because creating this target is only done in applications/badslam/CMakeLists.txt if `CMAKE_CUDA_COMPILER` has been set. You could try outputting the value of this to check. I don't think that setting `CMAKE_CUDA_FLAGS` will solve the issue, but I'm not sure which CMake variable has to be set manually to make it find CUDA. Maybe try manually setting `CMAKE_CUDA_COMPILER` to the path of the `nvcc` program (in my case: `/usr/local/cuda-10.1/bin/nvcc`)?

AbbyRosewood, 2019-11-28:
Thank you my friend. I've now able to find CUDA. But I still could not build it.
In the terminal it shows 
collect2: error: ld returned 1 exit status
make[3]: *** [applications/badslam/badslam] Error 1
make[2]: *** [applications/badslam/CMakeFiles/badslam.dir/all] Error 2
make[1]: *** [applications/badslam/CMakeFiles/badslam.dir/rule] Error 2
make: *** [badslam] Error 2

errors like"../../liblibvis.so: undefined reference to `__glewDetachShader'" occurred yet all the required packages was successfully found.

AbbyRosewood, 2019-11-29:
Sorry to bother but I still haven't solved my problem. Could the mismatch of the Cmake version cause this error? I'm now using cmake-3.16

libbadslam_baselib.a(pose_graph_optimizer.cc.o): In function `g2o::LinearSolverCSparse<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::solve(g2o::SparseBlockMatrix<Eigen::Matrix<double, 6, 6, 0, 6, 6> > const&, double*, double*)':
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:126: undefined reference to `g2o::csparse_extension::cs_cholsolsymb(cs_di_sparse const*, double*, cs_di_symbolic const*, double*, int*)'
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:130: undefined reference to `g2o::csparse_extension::writeCs2Octave(char const*, cs_di_sparse const*, bool)'
libbadslam_baselib.a(pose_graph_optimizer.cc.o): In function `g2o::LinearSolverCSparse<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::solveBlocks(double**&, g2o::SparseBlockMatrix<Eigen::Matrix<double, 6, 6, 0, 6, 6> > const&)':
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:171: undefined reference to `g2o::csparse_extension::cs_chol_workspace(cs_di_sparse const*, cs_di_symbolic const*, int*, double*)'
libbadslam_baselib.a(pose_graph_optimizer.cc.o): In function `g2o::LinearSolverCSparse<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::solvePattern(g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> >&, std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > > const&, g2o::SparseBlockMatrix<Eigen::Matrix<double, 6, 6, 0, 6, 6> > const&)':
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:208: undefined reference to `g2o::csparse_extension::cs_chol_workspace(cs_di_sparse const*, cs_di_symbolic const*, int*, double*)'
libbadslam_baselib.a(render_window.cc.o): In function `vis::BadSlamRenderWindow::SaveScreenshotImpl()':
/home/Abby/study/badslam/applications/badslam/src/badslam/render_window.cc:1001: undefined reference to `__glewGenFramebuffers'
/home/Abby/study/badslam/applications/badslam/src/badslam/render_window.cc:1002: undefined reference to `__glewBindFramebuffer'
/home/Abby/study/badslam/applications/badslam/src/badslam/render_window.cc:1016: undefined reference to `__glewFramebufferTexture2D'
/home/Abby/study/badslam/applications/badslam/src/badslam/render_window.cc:1020: undefined reference to `__glewCheckFramebufferStatus'
/home/Abby/study/badslam/applications/badslam/src/badslam/render_window.cc:1025: undefined reference to `__glewBindFramebuffer'
/home/Abby/study/badslam/applications/badslam/src/badslam/render_window.cc:1026: undefined reference to `__glewBindFramebuffer'
/home/Abby/study/badslam/applications/badslam/src/badslam/render_window.cc:1027: undefined reference to `__glewBlitFramebuffer'
/home/Abby/study/badslam/applications/badslam/src/badslam/render_window.cc:1030: undefined reference to `__glewBindFramebuffer'
/home/Abby/study/badslam/applications/badslam/src/badslam/render_window.cc:1039: undefined reference to `__glewBindFramebuffer'
/home/Abby/study/badslam/applications/badslam/src/badslam/render_window.cc:1041: undefined reference to `__glewDeleteFramebuffers'
libbadslam_baselib.a(render_window.cc.o): In function `vis::BadSlamRenderWindow::Initialize()':
/home/Abby/study/badslam/applications/badslam/src/badslam/render_window.cc:137: undefined reference to `glewInit'
/home/Abby/study/badslam/applications/badslam/src/badslam/render_window.cc:152: undefined reference to `__glewGenVertexArrays'
/home/Abby/study/badslam/applications/badslam/src/badslam/render_window.cc:153: undefined reference to `__glewBindVertexArray'
libbadslam_baselib.a(render_window.cc.o): In function `~PointCloudOpenGL':
/home/Abby/study/badslam/./libvis/src/libvis/point_cloud_opengl.h:86: undefined reference to `__glewDeleteBuffers'
/home/Abby/study/badslam/./libvis/src/libvis/point_cloud_opengl.h:86: undefined reference to `__glewDeleteBuffers'
libbadslam_baselib.a(render_window.cc.o): In function `vis::PointCloudOpenGL<vis::PointC<Eigen::Matrix<float, 3, 1, 0, 3, 1>, Eigen::Matrix<unsigned char, 3, 1, 0, 3, 1> > >::AllocateBuffer(unsigned long, unsigned int)':
/home/Abby/study/badslam/./libvis/src/libvis/point_cloud_opengl.h:94: undefined reference to `__glewGenBuffers'
/home/Abby/study/badslam/./libvis/src/libvis/point_cloud_opengl.h:98: undefined reference to `__glewBindBuffer'
/home/Abby/study/badslam/./libvis/src/libvis/point_cloud_opengl.h:99: undefined reference to `__glewBufferData'
libbadslam_baselib.a(render_window.cc.o): In function `vis::PointCloudOpenGL<vis::Point<Eigen::Matrix<float, 3, 1, 0, 3, 1> > >::TransferToGPU(unsigned long, unsigned long, float const*, unsigned int)':
/home/Abby/study/badslam/./libvis/src/libvis/point_cloud_opengl.h:113: undefined reference to `__glewBindBuffer'
/home/Abby/study/badslam/./libvis/src/libvis/point_cloud_opengl.h:114: undefined reference to `__glewBufferData'
/home/Abby/study/badslam/./libvis/src/libvis/point_cloud_opengl.h:109: undefined reference to `__glewGenBuffers'
libbadslam_baselib.a(render_window.cc.o): In function `vis::PointCloudOpenGL<vis::Point<Eigen::Matrix<float, 3, 1, 0, 3, 1> > >::SetAttributes(vis::ShaderProgramOpenGL*)':
/home/Abby/study/badslam/./libvis/src/libvis/point_cloud_opengl.h:185: undefined reference to `__glewBindBuffer'
libbadslam_baselib.a(render_window.cc.o): In function `vis::PointCloudOpenGL<vis::PointC<Eigen::Matrix<float, 3, 1, 0, 3, 1>, Eigen::Matrix<unsigned char, 3, 1, 0, 3, 1> > >::TransferToGPU(vis::PointCloud<vis::PointC<Eigen::Matrix<float, 3, 1,  [... cut ...]
../../liblibvis.so: undefined reference to `__glewLinkProgram'
../../liblibvis.so: undefined reference to `__glewEnableVertexAttribArray'
../../liblibvis.so: undefined reference to `__glewCreateShader'
../../liblibvis.so: undefined reference to `__glewDetachShader'
collect2: error: ld returned 1 exit status
make[3]: *** [applications/badslam/badslam] Error 1
make[2]: *** [applications/badslam/CMakeFiles/badslam.dir/all] Error 2
make[1]: *** [applications/badslam/CMakeFiles/badslam.dir/rule] Error 2
make: *** [badslam] Error 2

puzzlepaint, 2019-11-29:
There seem to be two problems here, a linking problem with GLEW and a linking problem with g2o. It seems to me that both are probably some kind of problem with the library setup, and not problems with BAD SLAM. I don't have concrete guesses for the cause of these issues. I think that I remember seeing the g2o issue myself in the past, but I don't remember what the problem was. For example, a possible problem could be if there are two versions of the library on your system, and you include the headers of one version while trying to link to the other. The problem is most likely unrelated to the CMake version you are using. By the way, if you enclose the error output with ``` then it looks nicer.

AbbyRosewood, 2019-12-25:
Sorry for replying you this late and thanks~ for your kind suggestion.

AbbyRosewood, 2020-01-06:
Hello~
I wonder how exactly can I save the point cloud and the key frames.
Thanks!

AbbyRosewood, 2020-01-06:
Hello~
I wonder how exactly can I save the point cloud and the key frames.
When I add the `--export_poses` or the `--export_point_cloud` command to the command line, it complains:
```
$ ./applications/badslam/badslam /home/Abby/Desktop/badslam-master/data/rgb
d_dataset_freiburg1_xyz/ --gui_run --export_point_cloud
time                            file:line     v| 
20:38:51.025              loguru.cpp:587   INFO| arguments: ./applications/badslam/badslam /home/Abby/Desktop/badslam-master/data/rgbd_dataset_freiburg1_xyz/ --gui_run --export_point_cloud
20:38:51.025              loguru.cpp:590   INFO| Current dir: /home/Abby/Desktop/badslam-master/build_RelWithDebInfo
20:38:51.025              loguru.cpp:592   INFO| stderr verbosity: 2
20:38:51.025              loguru.cpp:593   INFO| -----------------------------------
20:38:51.025                 main.cc:441   WARN| No auto-tuning file found -> using default parameters. GPU performance is thus probably slightly worse than it could be.
20:38:51.082                 main.cc:477    ERR| Trajectory path given, but loading a ground truth trajectory is not supported yet: --export_point_cloud
20:38:51.090              loguru.cpp:467   INFO| atexit
```
Thanks!

puzzlepaint, 2020-01-06:
You probably need to specify a file path as another command-line argument after `--export_point_cloud`, otherwise "--export_point_cloud" gets misinterpreted as a trajectory path.

Billy-0229, 2020-01-10:
Thanks a lot! Solved it.

closed 2020-04-14 by puzzlepaint