ETH3D/badslam #37: DUtils/DUtils.h: No such file or directory

Billy-0229, 2019-12-25:
Hi, I followed the steps in the `readme.` However, it encounters error as follows:

```
/home/B0229/Desktop/study/badslam-master/applications/badslam/third_party/DBoW2/src/FBrief.cpp:14:29: fatal error: DVision/DVision.h: No such file or directory
 #include <DVision/DVision.h>
                             ^
compilation terminated.
[ 31%] Built target png
/home/B0229/Desktop/study/badslam-master/applications/badslam/third_party/DBoW2/src/FORB.cpp:16:27: fatal error: DUtils/DUtils.h: No such file or directory
 #include <DUtils/DUtils.h>
                           ^
compilation terminated.
[ 31%] Automatic MOC for target libvis
applications/badslam/third_party/DBoW2/CMakeFiles/DBoW2.dir/build.make:101: recipe for target 'applications/badslam/third_party/DBoW2/CMakeFiles/DBoW2.dir/src/FORB.cpp.o' failed
make[3]: *** [applications/badslam/third_party/DBoW2/CMakeFiles/DBoW2.dir/src/FORB.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
applications/badslam/third_party/DBoW2/CMakeFiles/DBoW2.dir/build.make:88: recipe for target 'applications/badslam/third_party/DBoW2/CMakeFiles/DBoW2.dir/src/FBrief.cpp.o' failed
make[3]: *** [applications/badslam/third_party/DBoW2/CMakeFiles/DBoW2.dir/src/FBrief.cpp.o] Error 1
```

It seems DUtils/DUtils.h and DVision/DVision.h are missing

Any suggestion to fix it?

Thanks~

Billy-0229, 2019-12-25:
Here is a new one though:
```
/home/B0229/Desktop/study/badslam-master/applications/badslam/src/badslam/gui_keyframe_dialog.cc: In constructor ‘vis::KeyframeDialog::KeyframeDialog(std::atomic<long unsigned int>*, int, const std::shared_ptr<vis::Keyframe>&, const vis::BadSlamConfig&, vis::BadSlam*, const std::shared_ptr<vis::BadSlamRenderWindow>&, QWidget*)’:
/home/B0229/Desktop/study/badslam-master/applications/badslam/src/badslam/gui_keyframe_dialog.cc:170:79: error: ‘half’ in namespace ‘Eigen’ does not name a type
         radius_image_float(x, y) = static_cast<float>(reinterpret_cast<Eigen::half&>(radius_image(x, y)));
                                                                               ^
/home/B0229/Desktop/study/badslam-master/applications/badslam/src/badslam/gui_keyframe_dialog.cc:170:83: error: expected ‘>’ before ‘&’ token
         radius_image_float(x, y) = static_cast<float>(reinterpret_cast<Eigen::half&>(radius_image(x, y)));
                                                                                   ^
/home/B0229/Desktop/study/badslam-master/applications/badslam/src/badslam/gui_keyframe_dialog.cc:170:83: error: expected ‘(’ before ‘&’ token
/home/B0229/Desktop/study/badslam-master/applications/badslam/src/badslam/gui_keyframe_dialog.cc:170:84: error: expected primary-expression before ‘>’ token
         radius_image_float(x, y) = static_cast<float>(reinterpret_cast<Eigen::half&>(radius_image(x, y)));
                                                                                    ^
/home/B0229/Desktop/study/badslam-master/applications/badslam/src/badslam/gui_keyframe_dialog.cc:170:106: error: expected ‘)’ before ‘;’ token
         radius_image_float(x, y) = static_cast<float>(reinterpret_cast<Eigen::half&>(radius_image(x, y)));
                                                                                                          ^
applications/badslam/CMakeFiles/badslam_baselib.dir/build.make:225: recipe for target 'applications/badslam/CMakeFiles/badslam_baselib.dir/src/badslam/gui_keyframe_dialog.cc.o' failed
make[3]: *** [applications/badslam/CMakeFiles/badslam_baselib.dir/src/badslam/gui_keyframe_dialog.cc.o] Error 1
make[3]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:831: recipe for target 'applications/badslam/CMakeFiles/badslam_baselib.dir/all' failed
make[2]: *** [applications/badslam/CMakeFiles/badslam_baselib.dir/all] Error 2
CMakeFiles/Makefile2:774: recipe for target 'applications/badslam/CMakeFiles/badslam.dir/rule' failed
make[1]: *** [applications/badslam/CMakeFiles/badslam.dir/rule] Error 2
Makefile:435: recipe for target 'badslam' failed
make: *** [badslam] Error 2
```

I

puzzlepaint, 2019-12-28:
I believe that `Eigen::half` was added to Eigen relatively recently. Updating Eigen will probably fix this error.

Billy-0229, 2019-12-30:
THANKS, solved it~

puzzlepaint, 2019-12-30:
Which GPU do you use? I haven't seen this error before. Maybe it actually runs out of memory. If so, you could try to use the `--max_surfel_count` parameter to reduce the initial memory use, for example by setting it to something like 2 million instead of the default of 25 million.

Billy-0229, 2019-12-31:
I'm using `NVIDIA Corporation GM107GL [Quadro K620] (rev a2)`. Limiting the maximum number of the surfels actually worked. Then I ran into another problem which looks similar to #30 so I also ran it with gdb and it shows this:

```
gdb --ex run --args ./applications/badslam/badslam /home/B0229/Desktop/study/bad
slam-master/data/rgbd_dataset_freiburg1_xyz/ --gui_run
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./applications/badslam/badslam...done.
Starting program: /home/B0229/Desktop/study/badslam-master/build_RelWithDebInfo/applications/badslam/badslam /home/B0229/Desktop/study/badslam-master/data/rgbd_dataset_freiburg1_xyz/ --gui_run
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
time                            file:line     v| 
09:59:22.823              loguru.cpp:587   INFO| arguments: /home/B0229/Desktop/study/badslam-master/build_RelWithDebInfo/applications/badslam/badslam /home/B0229/Desktop/study/badslam-master/data/rgbd_dataset_freiburg1_xyz/ --gui_run
09:59:22.823              loguru.cpp:590   INFO| Current dir: /home/B0229/Desktop/study/badslam-master/build_RelWithDebInfo
09:59:22.823              loguru.cpp:592   INFO| stderr verbosity: 2
09:59:22.823              loguru.cpp:593   INFO| -----------------------------------
09:59:22.823                 main.cc:441   WARN| No auto-tuning file found -> using default parameters. GPU performance is thus probably slightly worse than it could be.
[New Thread 0x7fffe6302700 (LWP 11489)]
[New Thread 0x7fffdbb56700 (LWP 11490)]
[New Thread 0x7fffdb355700 (LWP 11491)]
[New Thread 0x7fffda947700 (LWP 11492)]
[New Thread 0x7fffd9f2a700 (LWP 11493)]
[New Thread 0x7fffc8931700 (LWP 11497)]
09:59:23.283      gui_main_window.cc:1600  INFO| Read dataset with 792 frames
[New Thread 0x7fffbb69a700 (LWP 11498)]
[New Thread 0x7fffbae99700 (LWP 11499)]
[New Thread 0x7fffba698700 (LWP 11500)]
[New Thread 0x7fffb3fff700 (LWP 11504)]

Thread 7 "badslam" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc8931700 (LWP 11497)]
0x00007ffff7bc67fd in boost::filesystem::detail::status(boost::filesystem::path const&, boost::system::error_code*) ()
   from /usr/local/lib/libboost_filesystem.so.1.66.0
(gdb) bt
#0  0x00007ffff7bc67fd in boost::filesystem::detail::status(boost::filesystem::path const&, boost::system::error_code*) ()
   from /usr/local/lib/libboost_filesystem.so.1.66.0
#1  0x0000000000436157 in boost::filesystem::exists (p=...)
    at /usr/local/include/boost/filesystem/operations.hpp:404
#2  vis::BadSlam::BadSlam (this=0x7fffc0507dd0, config=..., 
    rgbd_video=0x7fffffffb2b0, render_window=
    std::shared_ptr (count 5, weak 0) 0x1ad4a60, 
    opengl_context=<optimized out>)
    at //home/B0229/Desktop/study/badslam-master/applications/badslam/src/badslam/bad_slam.cc:141
#3  0x000000000046e141 in vis::MainWindow::WorkerThreadMain (
    this=0x7fffffffb1d0)
    at //home/B0229/Desktop/study/badslam-master/applications/badslam/src/badslam/gui_main_window.cc:1719
#4  0x00007ffff3110c80 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff4ed96ba in start_thread (arg=0x7fffc8931700)
    at pthread_create.c:333
#6  0x00007ffff287641d in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

It appeared to be a problem of my boost1.66. Should I change the version of this package?

Billy-0229, 2019-12-31:
I haven't figured out how I did it but I think I've solved it. The `boost1.66` is compatible with your program.

closed 2019-12-31 by Billy-0229

camrazor, 2020-02-11:
Can you please write how did you solve the problem with DVision.h and DUtils.h ?
I have installed libdlib-dev but cannot find these files in /usr/include/dlib
BR,
> Hi, I followed the steps in the `readme.` However, it encounters error as follows:
>
> ```
> /home/B0229/Desktop/study/badslam-master/applications/badslam/third_party/DBoW2/src/FBrief.cpp:14:29: fatal [...]

tkircher, 2020-04-09:
@camrazor There is a big difference between DLib and dlib. Yeah, pretty confusing. This project uses the former, which is not available on Ubuntu or Debian, can be found at:

 https://github.com/davisking/dlib

Unfortunately, it's been obsolete for years and is not recommended. I managed to get it to compile, but only after updating it from OpenCV 2. For instance, IplImage has not worked for ages and has to be replaced with Mat.

dianaleeg, 2021-08-13:
Anyone able to share how they moved past this? I am getting the same error regarding  **fatal error: DUtils/DUtils.h: No such file or directory #include <DUtils/DUtils.h>** and **fatal error: DVision/DVision.h: No such file or directory #include <DVision/DVision.h>**

 I have correct DLib installed (https://github.com/dorian3D/dlib) and I have environment variables pointing to DLib's include directory.