ETH3D/badslam #39: Privide example auto_tuning_result.txt

slovak194, 2020-02-05:
It would be great if example auto_tuning_result.txt will be included in the repository.
For now, it is hard to guess what to do if badslam crashes with **Cuda Error: out of memory** under the auto_tune_parameters.py

Many thanks!

puzzlepaint, 2020-02-05:
Does it crash immediately or after a while? In the former case, maybe try to reduce the `--max_surfel_count` parameter. In the latter case, maybe choose a smaller dataset for tuning. In any case, below is an example content of auto_tuning_result.txt (might be from an older version though). Note that auto-tuning is optional in case the program runs without it.

```
CalibrateDepthAndTransformColorToDepthCUDAKernel 32 16 0.0020464
UpdateSurfelPositionAndDescriptorCUDAKernel 1024 1 0.000597184
SetSurfelInactiveKernel 1024 1 0.000243616
ComputeBrightnessKernel 32 32 0.00495389
DownsampleImagesCUDAKernel 32 16 0.0135292
BilateralFilteringAndDepthCutoffCUDAKernel 8 16 0.00939517
ComputeNormalsCUDAKernel 16 16 0.00345818
DetermineSupportingSurfelsCUDAKernel 1024 1 0.00266832
CalibrateDepthCUDAKernel 32 16 0.00190826
FilterNewSurfelsCUDAKernel 1024 1 0.00015056
ComputeMinMaxPointRadiusCUDAKernel 32 16 0.00287501
AccumulatePoseEstimationCoeffsCUDAKernel 128 1 0.0372465
CompactSurfelsCUDAKernel 128 1 0.00016896
ResetSurfelAccumForSurfelDeletionAndRadiusUpdateCUDAKernel<true> 1024 1 5.2576e-05
DetermineActiveSurfelsKernel 1024 1 0.00117299
ResetSurfelForColorAssignmentKernel 1024 1 5.1104e-05
CreateSurfelsForKeyframeCUDACreationAppendKernel 8 16 0.000590368
CountObservationsAndFreeSpaceViolationsCUDAKernel<true> 1024 1 0.000849728
AccumulateColorObservationsCUDAKernel 128 1 0.00112176
CreateSurfelsForKeyframeCUDAWriteFreeSpotListKernel 128 1 9.3184e-05
ComputeCostAndResidualCountFromImagesCUDAKernel_GradientXY 8 16 0.019852
ComputeMinMaxDepthCUDAKernel 32 16 0.000247232
FlagInvalidSurfelsInAccum2CUDAKernel 256 1 0.000157344
AccumulateSurfelPositionAndDescriptorOptimizationCoeffsCUDAKernel<true> 128 1 0.0101246
UpdateSurfelNormalCUDAKernel 1024 1 0.000245408
ComputePointRadiiAndRemoveIsolatedPixelsCUDAKernel 32 16 0.00229155
ResetSurfelAccumCUDAKernel 1024 1 0.000274784
UpdateSurfelVertexBufferCUDAKernel<TEMPLATE_ARGUMENTS(false, false)> 32 1.0
AssignColorsCUDAKernel 1024 1 4.8416e-05
WriteNewSurfelIndexAndInitializeObservationsCUDAKernel 128 1 0.000325056
CountObservationsForNewSurfelsCUDAKernel 1024 1 0.00154256
ResetSurfelAccum0to3CUDAKernel 1024 1 0.000220832
AccumulatePoseEstimationCoeffsFromImagesCUDAKernel_GradientXY 8 16 0.203002
CreateSurfelsForKeyframeCUDASerializingKernel 32 32 0.00089648
AccumulateSurfelNormalOptimizationCoeffsCUDAKernel 1024 1 0.0037904
MarkDeletedSurfelsCUDAKernel 1024 1 5.7856e-05

```

slovak194, 2020-02-06:
>> reduce the --max_surfel_count parameter
Thanks, it helps!

closed 2020-02-06 by slovak194