ETH3D/badslam #40: Question: Usage with Azure Kinect

brookman, 2020-02-07:
Can the software in the current state somehow be used together with the Azure Kinect (k4a) device?
I have tried it with release 1.0.1 under Ubuntu 18.04 with no success. Or is that only possible when building from source? I am also struggling with compiling it or rather getting all the dependencies installed. More detailed instructions on how to install and configure everything would be tremendously helpful.

puzzlepaint, 2020-02-07:
Due to open licensing questions respectively nobody who owns a K4A device being willing to do the work to work around them, there is indeed no binary release that is compiled with K4A support. However, it should work when compiled from source.

I agree that it would be nice to have more detailed building instructions (for the latest Ubuntu LTS release).

brookman, 2020-02-08:
Thanks for the reply. I got it running (except for loop closure) after compiling it from source. I can confirm that it works with CUDA 10.2 as well.
I haven't got much experience with building C++ projects. So building and installing the dependencies took half a day. It might be an idea to provide a script or Dockerfile to simplify this process.

closed 2020-02-08 by brookman

puzzlepaint, 2020-04-14:
For anyone else possibly reading this: There is a pre-compiled Windows binary with support for the Azure Kinect now.