ETH3D/badslam #43: FATAL_ERROR "CMake did not find CUDA, which is required to build this program."

Young532042725, 2020-02-21:
hi, friends. Today, i build it on a server, named aaaa, my personal account name is yjs. As is shown below. The aaaa is already installed CUDA. in aaaa  run "env | grep "cuda" " has shown. But the CMakeLists.txt indicate FATAL_ERROR "CMake did not find CUDA, which is required to build this program.".  As is shown below.  can you give me some help? thanks!!!
[image]

puzzlepaint, 2020-02-21:
Would it work to run the following?

```
cmake \
    -DCMAKE_BUILD_TYPE=RelWithDebInfo \
    -DCMAKE_CUDA_FLAGS="-arch=(the_appropriate_architecture(s))" \
    -DCMAKE_CUDA_COMPILER=/usr/local/cuda-10.0/bin/nvcc \
    ..
```

Or alternatively to `-DCMAKE_CUDA_COMPILER`, try to set `CUDA_TOOLKIT_ROOT_DIR` to `/usr/local/cuda-10.0`. Or put `nvcc` into the `PATH` environment variable.

Young532042725, 2020-02-24:
@puzzlepaint  hello,friend. i have successfully compiled "badslam&&badslam_test". But when i run "./applications/badslam/badslam_test". some error occurs,as shown below. can you give me some advice? thank you!! it is tested on CUDA10.0 &&&& GTX 1070 !!
[image]

Young532042725, 2020-02-24:
@puzzlepaint 
[image]

puzzlepaint, 2020-03-01:
See #44.

closed 2020-03-01 by puzzlepaint

PyojinKim, 2020-10-16:
> @puzzlepaint hello,friend. i have successfully compiled "badslam&&badslam_test". But when i run "./applications/badslam/badslam_test". some error occurs,as shown below. can you give me some advice? thank you!! it is tested on CUDA10.0 &&&& GTX 1070 !!
> [image]

> @puzzlepaint hello,friend. i have successfully compiled "badslam&&badslam_test". But when i run "./applications/badslam/badslam_test". some error occurs,as shown below. can you give me some advice? thank you!! it is tested on CUDA10.0 &&&& GTX 1070 !!
> [image]

Hi @jsYangCode , Could I ask how to solve "CMake did not find CUDA, which is required to build this program." this error? I also installed CUDA, but see the same error message when I did CMake.