ETH3D/badslam #44: QXcbConnection: Could not connect to display

Young532042725, 2020-02-24:
hello,friends. i have successfully compiled "badslam&&badslam_test". But when i run "./applications/badslam/badslam_test". some error occurs,as shown below.   can you give me some advice? thank you!!  it is tested on CUDA10.0 &&&& GTX 1070 !!
[image]

Young532042725, 2020-02-24:
[image]

Young532042725, 2020-02-24:
[image]

Young532042725, 2020-02-26:
@puzzlepaint hello friend, would you please give me some help according to my issues, thank you !

puzzlepaint, 2020-03-01:
I would suggest to try googling the error and trying some of the solutions found in this way. This error seems to occur because Qt cannot find the display, and I think it is most likely not an error in badslam.