ETH3D/badslam #46: read distortion parameter

zhaozhongch, 2020-03-17:
I am trying to run badslam on TUM rgbd dataset. In the `readme` I found that the `calibration.txt` only has pinhole camera model without any distortion parameter.
Will the code work if I simply add some distortion parameter after camera intrinsics? Like
```
fx fy cx cy d1 d2 d3 d4 d5
```
In `rgbd_video_io_tum_dataset.h` I didn't find any piece of code that can read distortion parameter.
Also in the readme seems for tum rgbd I don't need to input depth calibration file?
I really appreciate your answer!

puzzlepaint, 2020-03-19:
The application only supports the pinhole camera model, so the images need to be pre-undistorted. Simply adding distortion parameters afterwards will of course not work. However, I believe that those are not necessary for the TUM RGB-D datasets.

zhaozhongch, 2020-03-19:
Thanks for the answer. Here is another concern.
If I have another rgbd dataset in the future, whose depth is aligned with the rgb. After I undistort the rgb image, should I use the same parameter to undistort the depth? My concern is that after undistortion, the rgb's pixel may not be able to align with the depth.

puzzlepaint, 2020-03-23:
Transforming depth images is always a bit tricky, since it is generally not easily possible to *correctly* interpolate depth values in depth images. Two alternatives I would suggest are:

* Perform the depth image undistortion while attempting to detect depth discontinuities in the depth image. If there is no discontinuity at a given target pixel, the depth could be interpolated by assuming that the surface is planar there; if a discontinuity is estimated to be at a pixel, then no depth value should be assigned to this pixel since its true depth is unclear. Unfortunately, there is probably no fail-safe way to detect the discontinuities, but potentially a simple check on the angle of a plane fitted to neighboring pixel values to the camera might work sufficiently well.
* Try to avoid the undistortion, if possible, by implementing the corresponding original camera model in the application instead of undistorting the images to the pinhole camera model, in case this does not make it too slow.

closed 2020-03-23 by zhaozhongch