ETH3D/badslam #51: Shipped DBoW2 package compilation error

sqn175, 2020-07-13:
Hi, 

Thanks for your great work!
I'm trying to compile badslam, however some errors that may relate to the shipped **DBoW2** package, appear as follows: 
`/home/qin/Documents/codes/3rdparty/bin_3rdparty/DLib/include/DLib/../DVision/BRIEF256.h: In member function ‘void DVision::BRIEF_t<Bits>::compute(const cv::Mat&, const std::vector<cv::KeyPoint>&, std::vector<std::bitset<Bits> >&, bool) const’:
/home/qin/Documents/codes/3rdparty/bin_3rdparty/DLib/include/DLib/../DVision/BRIEF256.h:207:11: error: ‘cvtColor’ is not a member of ‘cv’
       cv::cvtColor(image, aux, cv::COLOR_RGB2GRAY);`

Note: 
1. I have pre-compiled DLib and installed the compiled stuff to the directory '/home/qin/Documents/codes/3rdparty/bin_3rdparty/DLib' using '-DCMAKE_INSTALL_PREFIX';
2. My opencv version is 3.2.0 as shown below:
`$ apt list -a libopencv-dev
Listing... Done
libopencv-dev/bionic-updates,bionic-security,now 3.2.0+dfsg-4ubuntu0.1 amd64 [installed,automatic]
libopencv-dev/bionic 3.2.0+dfsg-4build2 amd64
`

Another way to reproduce the error:
1. Copy the shipped directory DBoW2 to other place, e.g., /home/DoW2;
2. $ cd /home/DoW2 && mkdir build && cd build && cmake .. && make -j8

The same errors will occur. 

Is there any solution to solve this compilation problem? Many thanks!

Best regards,
Qin Shi

puzzlepaint, 2020-07-14:
There was just a modification to the DBoW2 files related to BRIEF, maybe this adds some missing include: #50 

If that does not fix it, have you tried updating the shipped version of DBoW2 to the latest official version? It seems like some inconsistency arose from the DLib/DBoW2/OpenCV versions that are present here.

37, 2020-07-19:
Just wanted to add that I get the same issue. I've copied the new DBoW files across, but kept the included CMakeLists and I can progress in compiling. Still getting other compilation issues however I'll likely make separate tickets for those.

sqn175, 2020-07-24:
> There was just a modification to the DBoW2 files related to BRIEF, maybe this adds some missing include: #50
>
> If that does not fix it, have you tried updating the shipped version of DBoW2 to the [...]

Using the DLib commit b6c28fb will solve the DBoW2 compilation problem. However, another compilation error that may relate to the flann arise as follows:
[image]

Is there any possible solution? Many thanks!

puzzlepaint, 2020-07-29:
It appears as if some dependency on your system requires OpenMP, but it is not linked in. I would try to link OpenMP explicitly (`-fopenmp` compiler flag) and see whether that fixes it.

sqn175, 2020-12-04:
> dependency

Your are right. After revisiting this issue, I found that I'm compiling g2o with "-DG2O_USE_OPENMP=ON". Compiling with "OFF" flag will solve the problem.

closed 2020-12-04 by sqn175