ETH3D/badslam #52: Compilation error

37, 2020-07-19:
Hi there. I'm having some trouble compiling on ubuntu 18.04. 

I've installed all of the dependencies and after following the guidance in #51 (downloading and copying Dorian3d's DBoW2 [3924753](https://github.com/dorian3d/DBoW2)  and moving it to the location of the included DBoW2 package I encounter new issues: 

```

$ make -j2 badslam
...
compiling
...

~/badslam/applications/badslam/src/badslam/loop_detector.cc: In constructor ‘vis::LoopDetector::LoopDetector(const string&, const string&, int, int, float, int, int, int, float, bool)’:
~/badslam/applications/badslam/src/badslam/loop_detector.cc:179:47: error: invalid new-expression of abstract class type ‘vis::BriefExtractor’
   extractor_.reset(new TExtractor(pattern_path));
                                               ^
In file included from ~/badslam/applications/badslam/src/badslam/loop_detector.cc:80:0:
~/badslam/applications/badslam/src/badslam/loop_detector.h:125:7: note:   because the following virtual functions are pure within ‘vis::BriefExtractor’:
 class BriefExtractor: public FeatureExtractor<FBrief::TDescriptor>
       ^~~~~~~~~~~~~~
~/badslam/applications/badslam/src/badslam/loop_detector.h:118:16: note: 	void vis::FeatureExtractor<TDescriptor>::operator()(const cv::Mat&, std::vector<cv::KeyPoint>&, std::vector<_Tp>&) const [with TDescriptor = std::bitset<256ul>]
   virtual void operator()(
                ^~~~~~~~
~/badslam/applications/badslam/src/badslam/loop_detector.cc: In member function ‘bool vis::LoopDetector::DetectLoop(const cv::Mat_<unsigned char>&, const std::shared_ptr<vis::Image<short unsigned int> >&, DLoopDetector::DetectionResult*, std::vector<cv::KeyPoint>*, std::vector<cv::KeyPoint>*, std::vector<cv::KeyPoint>*)’:
~/badslam/applications/badslam/src/badslam/loop_detector.cc:742:42: error: no match for call to ‘(vis::BriefExtractor) (const cv::Mat_<unsigned char>&, std::vector<cv::KeyPoint>&, std::vector<std::bitset<256ul> >&)’
   (*extractor_)(image, *keys, descriptors);
                                          ^
~/badslam/applications/badslam/src/badslam/loop_detector.cc:118:6: note: candidate: virtual void vis::BriefExtractor::operator()(const cv::Mat&, std::vector<cv::KeyPoint>&, std::vector<boost::dynamic_bitset<> >&) const
 void BriefExtractor::operator() (
      ^~~~~~~~~~~~~~
~/badslam/applications/badslam/src/badslam/loop_detector.cc:118:6: note:   no known conversion for argument 3 from ‘std::vector<std::bitset<256ul> >’ to ‘std::vector<boost::dynamic_bitset<> >&’
[ 89%] Building CXX object applications/badslam/CMakeFiles/badslam_baselib.dir/src/badslam/pairwise_frame_tracking.cc.o
In file included from ~/badslam/applications/badslam/third_party/DLoopDetector/DLoopDetector.h:54:0,
                 from ~/badslam/applications/badslam/src/badslam/loop_detector.h:85,
                 from ~/badslam/applications/badslam/src/badslam/loop_detector.cc:80:
~/badslam/applications/badslam/third_party/DLoopDetector/TemplatedLoopDetector.h: In instantiation of ‘void DLoopDetector::TemplatedLoopDetector<TDescriptor, F>::removeImage(int) [with TDescriptor = std::bitset<256ul>; F = DBoW2::FBrief]’:
~/badslam/applications/badslam/src/badslam/loop_detector.cc:714:28:   required from here
~/badslam/applications/badslam/third_party/DLoopDetector/TemplatedLoopDetector.h:886:15: error: ‘class DBoW2::TemplatedDatabase<std::bitset<256ul>, DBoW2::FBrief>’ has no member named ‘remove’
   m_database->remove(entry_id, bowvec);
   ~~~~~~~~~~~~^~~~~~
applications/badslam/CMakeFiles/badslam_baselib.dir/build.make:628: recipe for target 'applications/badslam/CMakeFiles/badslam_baselib.dir/src/badslam/loop_detector.cc.o' failed
make[3]: *** [applications/badslam/CMakeFiles/badslam_baselib.dir/src/badslam/loop_detector.cc.o] Error 1
make[3]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:952: recipe for target 'applications/badslam/CMakeFiles/badslam_baselib.dir/all' failed
make[2]: *** [applications/badslam/CMakeFiles/badslam_baselib.dir/all] Error 2
CMakeFiles/Makefile2:880: recipe for target 'applications/badslam/CMakeFiles/badslam.dir/rule' failed
make[1]: *** [applications/badslam/CMakeFiles/badslam.dir/rule] Error 2
Makefile:435: recipe for target 'badslam' failed
make: *** [badslam] Error 2

```

I'm not exactly sure what to do to proceed from here. 

Any advice?

puzzlepaint, 2020-07-20:
Okay, so the library update seems to have caused additional issues with the loop detection code. In addition, I forgot why I included copies of the DBoW2 and DLoopDetector libraries in the repository in the first place: Since I added a remove() function to the loop detector to remove keyframes.

* One way to proceed would be to try to adapt the loop detector code to the external changes, and either re-add the remove() function from the old code, or simply comment its use out (removing keyframes only happens if triggered via the GUI or if GPU memory is about to run out, however it is in a rather experimental state anyway, so it should not be used in practice).
* Another way could be to revert the library updates and use an old version of DLib that is compatible with the shipped libraries. On my Ubuntu 18.04 system, I used revision b6c28fb8c0e27e02a467f1ec8e0beed9a05f9433 of DLib.

37, 2020-07-22:
Thanks for that. I'll look into it tonight and will see if I'm able to get things running with either either approach - If I can re-implement remove() / removeImage successfully then I'll try to clean things up and submit a stable pull request. If not, I'll update here with more info :v: