ETH3D/badslam #53: Cuda Error : invalid augument

gongyeted, 2020-07-23:
Hello,friends. Sorry to bother you. I have successfully compiled "badslam&badslam_test", and badslam_test run successfully. But when i run "/applications/badslam/scripts/auto_tune_parameters.py", some error occurs as shown below, and I can't find solutions no matter how. Can you give me some advice? Thank you!! It is tested on CUDA10.0.30 &&&& GTX 1080Ti !!

Running: ./applications/badslam/badslam ../../datasets/BADSLAM_eth3d_slam_datasets/training/sfm_house_loop --sequential_ba --sequential_loop_detection --auto_tuning_iteration 0
time                            file:line     v|
20:11:15.291              loguru.cpp:587   INFO| arguments: ./applications/badslam/badslam ../../datasets/BADSLAM_eth3d_slam_datasets/training/sfm_house_loop --sequential_ba --sequential_loop_detection --auto_tuning_iteration 0
20:11:15.291              loguru.cpp:590   INFO| Current dir: /home/gy/SLAM/badslam/build_RelWithDebInfo
20:11:15.291              loguru.cpp:592   INFO| stderr verbosity: 2
20:11:15.291              loguru.cpp:593   INFO| -----------------------------------
20:11:17.518                 main.cc:529   INFO| Read dataset with 0 frames
Stack trace:
7       0x5609043f786a ./applications/badslam/badslam(+0x2b86a) [0x5609043f786a]
6       0x7f5d80503b97 __libc_start_main + 231
5       0x5609043f0291 ./applications/badslam/badslam(+0x24291) [0x5609043f0291]
4       0x5609043f91e7 ./applications/badslam/badslam(+0x2d1e7) [0x5609043f91e7]
3       0x560904409103 ./applications/badslam/badslam(+0x3d103) [0x560904409103]
2       0x56090457ac55 ./applications/badslam/badslam(+0x1aec55) [0x56090457ac55]
1       0x56090457ebbd vis::CUDABuffer<unsigned char>::CreateTextureObject(cudaTextureAddressMode, cudaTextureAddressMode, cudaTextureFilterMode, cudaTextureReadMode, bool, unsigned long long*) const + 797
0       0x7f5d82f4f163 loguru::StreamLogger::~StreamLogger() + 163
20:11:24.197       cuda_buffer_inl.h:213   **FATL| Cuda Error: invalid argument**
Program call failed (return code: -6)

And when I run "./applications/badslam/badslam" another error as shown below. Have no clue why. Wish may have your advice!!

time                            file:line     v|
20:22:48.548              loguru.cpp:587   INFO| arguments: applications/badslam/badslam
20:22:48.548              loguru.cpp:590   INFO| Current dir: /home/gy/SLAM/badslam/build_RelWithDebInfo
20:22:48.549              loguru.cpp:592   INFO| stderr verbosity: 2
20:22:48.549              loguru.cpp:593   INFO| -----------------------------------
20:22:48.549                 main.cc:441   WARN| No auto-tuning file found -> using default parameters. GPU performance is thus probably slightly worse than it could be.
20:23:06.211      gui_main_window.cc:1600  INFO| Read dataset with 0 frames
20:23:06.274 shader_program_opengl.c:98     **ERR| GL Shader Compilation Error: 0:1(10): error: GLSL 1.50 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES**

Stack trace:
25      0x55d76edcc86a applications/badslam/badslam(+0x2b86a) [0x55d76edcc86a]
24      0x7f9c5781fb97 __libc_start_main + 231
23      0x55d76edc5291 applications/badslam/badslam(+0x24291) [0x55d76edc5291]
22      0x55d76edce358 applications/badslam/badslam(+0x2d358) [0x55d76edce358]
21      0x55d76ee224c4 applications/badslam/badslam(+0x814c4) [0x55d76ee224c4]
20      0x7f9c59980beb QWidget::setVisible(bool) + 491
19      0x7f9c5997deef QWidgetPrivate::show_helper() + 79
18      0x7f9c5997de40 QWidgetPrivate::showChildren(bool) + 224
17      0x7f9c59980beb QWidget::setVisible(bool) + 491
16      0x7f9c5997deef QWidgetPrivate::show_helper() + 79
15      0x7f9c5997de40 QWidgetPrivate::showChildren(bool) + 224
14      0x7f9c59980beb QWidget::setVisible(bool) + 491
13      0x7f9c5997deef QWidgetPrivate::show_helper() + 79
12      0x7f9c5997de40 QWidgetPrivate::showChildren(bool) + 224
11      0x7f9c59980beb QWidget::setVisible(bool) + 491
10      0x7f9c5997ded3 QWidgetPrivate::show_helper() + 51
9       0x7f9c5997a0fb QWidgetPrivate::sendPendingMoveAndResizeEvents(bool, bool) + 427
8       0x7f9c58bc49c8 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 280
7       0x7f9c5994a104 QApplication::notify(QObject*, QEvent*) + 676
6       0x7f9c5994283c QApplicationPrivate::notify_helper(QObject*, QEvent*) + 156
5       0x7f9c59981bf2 QWidget::event(QEvent*) + 3490
4       0x7f9c599a1c72 QOpenGLWidget::resizeEvent(QResizeEvent*) + 66
3       0x7f9c599a15fd /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1b35fd) [0x7f9c599a15fd]
2       0x7f9c5a51dc52 vis::RenderWidgetOpenGL::initializeGL() + 34
1       0x55d76ef6c0de applications/badslam/badslam(+0x1cb0de) [0x55d76ef6c0de]
0       0x55d76ef6bd5e applications/badslam/badslam(+0x1cad5e) [0x55d76ef6bd5e]
20:23:06.274        render_window.cc:1357  FATL| CHECK FAILED:  !!(splat_program_.AttachShader( "#version 150\n" "uniform mat4 u_model_view_projection_matrix;\n" "in vec4 in_position;\n" "in vec3 in_color;\n" "out vec3 var1_color;\n" "void main() {\n" "  var1_color = in_color;\n" "  gl_Position = u_model_view_projection_matrix * in_position;\n" "}\n", ShaderProgramOpenGL::ShaderType::kVertexShader))
Aborted (core dumped)

gongyeted, 2020-07-23:
@puzzlepaint wish your help!

puzzlepaint, 2020-07-23:
It seems weird that this simple shader fails to compile. Are you sure that the (proprietary) Nvidia graphics driver is installed and is working properly?

gongyeted, 2020-07-24:
> It seems weird that this simple shader fails to compile. Are you sure that the (proprietary) Nvidia graphics driver is installed and is working properly?

I think so. Information are shown below.

gy@cvrs10:~$ lspci |grep -i vga
04:00.0 VGA compatible controller: NVIDIA Corporation GP102 [GeForce GTX 1080 Ti] (rev a1)
05:00.0 VGA compatible controller: NVIDIA Corporation GP102 [GeForce GTX 1080 Ti] (rev a1)
06:00.0 VGA compatible controller: NVIDIA Corporation GP102 [GeForce GTX 1080 Ti] (rev a1)
07:00.0 VGA compatible controller: NVIDIA Corporation GP102 [GeForce GTX 1080 Ti] (rev a1)
08:00.0 VGA compatible controller: NVIDIA Corporation GP102 [GeForce GTX 1080 Ti] (rev a1)
0b:00.0 VGA compatible controller: NVIDIA Corporation GP102 [GeForce GTX 1080 Ti] (rev a1)
0c:00.0 VGA compatible controller: NVIDIA Corporation GP102 [GeForce GTX 1080 Ti] (rev a1)
0d:00.0 VGA compatible controller: NVIDIA Corporation GP102 [GeForce GTX 1080 Ti] (rev a1)
0e:00.0 VGA compatible controller: NVIDIA Corporation GP102 [GeForce GTX 1080 Ti] (rev a1)
0f:00.0 VGA compatible controller: NVIDIA Corporation GP102 [GeForce GTX 1080 Ti] (rev a1)
12:00.0 VGA compatible controller: ASPEED Technology, Inc. ASPEED Graphics Family (rev 30)

To check nvidia graphics driver version:

gy@cvrs10:~$ cat /proc/driver/nvidia/version
NVRM version: NVIDIA UNIX x86_64 Kernel Module  430.26  Tue Jun  4 17:40:52 CDT 2019
GCC version:  gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.11)
gy@cvrs10:~$ sudo dpkg --list |grep nvidia-*
[sudo] password for gy:
ii  nvidia-430                                 430.26-0ubuntu0~gpu16.04.1                          amd64        NVIDIA binary driver - version 430.26
ii  nvidia-opencl-icd-430                      430.26-0ubuntu0~gpu16.04.1                          amd64        NVIDIA OpenCL ICD
ii  nvidia-prime                               0.8.8.2                                             all          Tools to enable NVIDIA's Prime
ii  nvidia-settings                            418.56-0ubuntu0~gpu18.04.1                          amd64        Tool for configuring the NVIDIA graphics driver

To check if graphics working normally:

gy@cvrs10:~$ glxinfo |grep rendering
direct rendering: Yes

More information about graphics working status:

gy@cvrs10:~$ nvidia-smi
Fri Jul 24 16:15:10 2020
+-----------------------------------------------------------------------------+
| **NVIDIA-SMI 430.26       Driver Version: 430.26       CUDA Version: 10.2**     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 108...  On   | 00000000:04:00.0 Off |                  N/A |
| 30%   53C    P2    60W / 200W |   1169MiB / 11178MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   1  GeForce GTX 108...  On   | 00000000:05:00.0 Off |                  N/A |
| 49%   82C    P2    89W / 200W |   9891MiB / 11178MiB |     98%      Default |
+-------------------------------+----------------------+----------------------+
|   2  GeForce GTX 108...  On   | 00000000:06:00.0 Off |                  N/A |
| 50%   84C    P2    98W / 200W |  10445MiB / 11178MiB |     84%      Default |
+-------------------------------+----------------------+----------------------+
|   3  GeForce GTX 108...  On   | 00000000:07:00.0 Off |                  N/A |
| 50%   83C    P2    83W / 200W |  10441MiB / 11178MiB |     95%      Default |
+-------------------------------+----------------------+----------------------+
|   4  GeForce GTX 108...  On   | 00000000:08:00.0 Off |                  N/A |
| 44%   74C    P2    90W / 200W |  10441MiB / 11178MiB |     51%      Default |
+-------------------------------+----------------------+----------------------+
|   5  GeForce GTX 108...  On   | 00000000:0B:00.0 Off |                  N/A |
| 42%   71C    P2    78W / 200W |  10405MiB / 11178MiB |     27%      Default |
+-------------------------------+----------------------+----------------------+
|   6  GeForce GTX 108...  On   | 00000000:0C:00.0 Off |                  N/A |
| 49%   82C    P2    83W / 200W |  10475MiB / 11178MiB |     98%      Default |
+-------------------------------+----------------------+----------------------+
|   7  GeForce GTX 108...  On   | 00000000:0D:00.0 Off |                  N/A |
| 46%   76C    P2    75W / 200W |  10299MiB / 11178MiB |     27%      Default |
+-------------------------------+----------------------+----------------------+
|   8  GeForce GTX 108...  On   | 00000000:0E:00.0 Off |                  N/A |
| 23%   34C    P8     8W / 200W |   3613MiB / 11178MiB |      0%      Default |
[... 1 lines cut ...]
|   9  GeForce GTX 108...  On   | 00000000:0F:00.0 Off |                  N/A |
| 23%   38C    P8     8W / 200W |     10MiB / 11178MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
[... cut ...]
|    3      7052      C   .../anaconda3/envs/pointpillars/bin/python 10431MiB |
|    4      7053      C   .../anaconda3/envs/pointpillars/bin/python 10431MiB |
|    5      7054      C   .../anaconda3/envs/pointpillars/bin/python 10395MiB |
|    6      7055      C   .../anaconda3/envs/pointpillars/bin/python 10465MiB |
|    7      7056      C   .../anaconda3/envs/pointpillars/bin/python 10289MiB |
+-----------------------------------------------------------------------------+

It seems that everything is ok. puzzled.

puzzlepaint, 2020-07-29:
My guess would be that something (maybe some configuration on the system) makes the application not use the Nvidia GPU(s) correctly. There seems to be an "ASPEED" GPU in addition, maybe it is trying to use that one? This is just a wild guess though since I don't have experience with using multi-GPU systems. In case the system is a server without a display, then this would also seem like a likely candidate for such issues.

gongyeted, 2020-07-31:
> My guess would be that something (maybe some configuration on the system) makes the application not use the Nvidia GPU(s) correctly. There seems to be an "ASPEED" GPU in addition, maybe it is trying [...]

Thank you for your help. I have successfully run the badslam in anothor PC with one RTX2070. When I run bad_test, it shows 2 failed with error about 0.32. Nevertheless, it works well on most of your datasets. Thank you for your great work again.

closed 2020-09-24 by gongyeted