ETH3D/badslam #54: using cmake to compile in vs2019 + windows (64) + cuda10.1，Eigen function in kernel function of libvis_cuda lead to error

zhulz, 2020-07-24:
use cmake to compile in vs2019 + windows (64) + cuda10.1, in libvis_ CUDA's Cu file there has the eigen function and define. this markes that  libvis_ CUDA is failed to compile. i find that you provide the exe files in the windows ,   How do you compile it in windows ?
@puzzlepaint wish your help!

puzzlepaint, 2020-07-29:
It is possible that an Eigen include got introduced in a .cu file again after the Windows version was compiled. Which file is it? The include should be removed from that file to fix the problem.

zhulz, 2020-08-04:
When I reconfigured and then compiled, it passed. Thanks

zapaishchykova, 2020-09-13:
Hey, Im trying to compile on similar setup - vs19+cuda 10.2 + win 10 x64, and I get the following error:
```
Severity	Code	Description	Project	File	Line	Suppression State
Error		CMake Error at D:/Installed/VisualStudio2019/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:164 (message):
  Could NOT find Eigen3 (missing: EIGEN3_INCLUDE_DIR EIGEN3_VERSION_OK)
  (Required is at least version "2.91.0")		D:/Installed/VisualStudio2019/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake	164	
```
Can you please describe how did you reconfigure it? Many thanks!

zhulz, 2020-09-13:
the error gives you good tips.  that is , your eigen3 library  is not  included

zapaishchykova, 2020-09-13:
@zhulz so i tried to include this in the beginning of the Makefile
```
set(EIGEN3_INCLUDE_DIR "D:\\Installed\\eigen\\eigen-3.3.7\\eigen-3.3.7\\cmake")
include_directories(${EIGEN3_INCLUDE_DIR})
```
And it did not help, unfortunately :(

zhulz, 2020-09-13:
1. When you compile the code on windows. What it needs is cmakelist.txt not makefile.
2. Eigen library is not needed to compile,  if the path of the eigen is  xxx\\xxx\\Eigen\\Dense., you try to set set(EIGEN3_INCLUDE_DIR "xxx\\xxx"), the Eigen library is used by including "#include<Eigen/Dense>" in projection or solution. so you only set the path xxxx\\xxxx.

zapaishchykova, 2020-09-13:
Sorry my bad, i meant its in the cmakelist.txt. I did not understand what you mean by the 2nd point, can you clarify it?

zhulz, 2020-09-13:
for  example ,  the path of eigen library is "‪E:\TOOLE\badslam111\badslam-master\extlib\Eigen337\Eigen\Dense" in my computer.
[image]

the cmake-gui is configured as below: 
[image]

zapaishchykova, 2020-09-13:
I see, I tried to set also one level higher and issue is still the same. Did you succeed compiling it on Win with Cmake?

zhulz, 2020-09-13:
yes, i compiled it successfully.

zhulz, 2020-09-13:
you can paste how you configure it in cmake-gui and the path of the library.  maybe i can  find some problems  and  give you some helps.

zapaishchykova, 2020-09-13:
Thanks! Now I'm bothered by the boost configuration error:
```
CMake Error at D:/Installed/Cmake/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:165 (message):
  Could NOT find Boost (missing: serialization filesystem system) (found
  version "1.66.0")

```
Although I set it up D:\Download\boost_1_66_0
[image]
How to set up it correctly?

zhulz, 2020-09-13:
maybe this picture is useful!
[image]

zapaishchykova, 2020-09-13:
My problem is that i don't have a vc142-x64 folder inside boost_1_73:( Did you compile/install boost library too?

zhulz, 2020-09-14:
yes, i compile the boost library.  the boost library .The boost library can be compiled with vs17 or vs15. 

You can first refer to my configure figure and add the .lib at the corresponding position in cmake GUI. Although there is no v142-x64 library, other versions of the library may also work.

for example,  in the cmake-gui  listview , the  value of  "Boost_SERIALIZATION_LIBRARY_RELEASE "  also can be set as " boost_serialization-vc140-mt-x64-1_66.lib"  or other version. You can try it and see if it works.

meitiever, 2021-03-07:
I got this error when compiling with win 10 + CUDA 11 + vs2019 community.
[image]

zhulz, 2021-03-08:
From the error information, QT is not configured well ， some lib  have not added！

meitiever, 2021-03-08:
I have included all qt libs of 5.9.9 or 5.14.0.  And the error also pops up.