ETH3D/badslam #55: Is there any solution for the QT Thread Error?

talesoff, 2020-07-29:
I built BADSLAM on win10 with VS2017 and QT 5.14.2.  

I can run the application, but there was an error as follows: 

ASSERT failure in QCoreApplication::sendEvent: "Cannot send events to objects owned by a different thread. Current thread 0x0x1dfec82ca40. Receiver '' (of type 'QOffscreenSurface') was created in thread 0x0x1df82c46670", file kernel\qcoreapplication.cpp, line 573

If I ignore the error then application works, but slower than binary built version.

Is there any hint or patch for the issue?

ps. Thank you for sharing nice research result.

puzzlepaint, 2020-07-29:
I remember that during development, there were some issues with thread affinity of Qt-allocated OpenGL context objects (after switching to those from another version of the context allocation code that does not use Qt). So I would expect this issue to very likely come from the same area, which would be the windowless OpenGL context handling in [libvis/opengl_context_qt.cc](https://github.com/ETH3D/badslam/blob/master/libvis/src/libvis/opengl_context_qt.cc) (which deals with a QOffscreenSurface). This code sets the `Qt::AA_DontCheckOpenGLContextThreadAffinity` attribute to disable some thread affinity check. I guess that perhaps the new issue showed up due to the newer Qt version used by you. Unfortunately, I don't have a patch for this, but maybe the above helps to find the problem.

> If I ignore the error then application works, but slower than binary built version.

Did you compile the application and the dependencies in Release mode, and include your CUDA architecture in CMAKE_CUDA_FLAGS for badslam?

talesoff, 2020-08-05:
For the second issue, I successfully built the program. 

Thanks for your help. 

Besides, the release-built-program with vs2017 does not load Kinect's stream data. (There may be some mistakes.)
So I built the program with vs2019.