ETH3D/badslam #59: BAD SLAM benchmark results of ICL-NUIM datasets?

PyojinKim, 2020-11-27:
Hi,

I am just curious that are there any reported BAD SLAM benchmark results on the ICL-NUIM datasets?
When I run BAD SLAM on ICL-NUIM datasets with default parameters, it easily fails to estimate the 6-DoF motion of the camera.
It diverges so quickly and easily on most of the scenes, so I cannot evaluate the accuracy of BAD SLAM on ICL-NUIM datasets.

[image]

Is it normal for BAD SLAM to diverge in this way? like the above screenshot?
Or what did I do something wrong? parameter setting or CUDA block tuning?

Thanks,
Pyojin Kim.

puzzlepaint, 2020-11-29:
I don't remember having tried to run the system on these datasets, so I tried it now.

I downloaded all the "TUM RGB-D Compatible PNGs with noise" from https://www.doc.ic.ac.uk/~ahanda/VaFRIC/iclnuim.html . To make them run in badslam, I modified the existing `associations.txt` files to replace `depth` with `rgb` and vice versa, and saved them as `associated.txt`. I added the `calibration.txt` file with the following content: `481.20 480.00 319.50 239.50`. **It is important not to use a negative focal length here as the ICL-NUIM site would suggest** ( https://www.doc.ic.ac.uk/~ahanda/VaFRIC/codes.html ). The reason is that the code would need to actively support this in some places, for example, in normal vector computation, by ensuring that the normal vectors don't get flipped by the unexpectedly inverted y-axis. However, I don't have any plans to find and adjust all of these places just in order to support this quirk of this dataset. If anything, I'd rather make badslam exit with an error if it encounters a negative focal length. So, a sane positive value should be used (by inverting the negative value) and care needs to be taken that the estimated trajectory might need to be flipped afterwards to compare it with the provided ground truth.

I left all of badslam's parameters at their default with the exception of the maximum depth setting in the Depth Preprocessing settings, which I increased to 5. This is because the virtual camera simulated in this dataset seems to have very little depth noise, so there is no need to discard large depth values. Probably the value should even be set to anything very large (such as 9999) to simply disable this altogether.

I did not do any CUDA parameter tuning and did not set up loop detection.

With this, the qualitative success / failure results were as follows:

* `living_room_traj0n_frei_png`: fails
* `living_room_traj1n_frei_png`: works
* `living_room_traj2n_frei_png`: works
* `living_room_traj3n_frei_png`: fails
* `traj0n_frei_png` fails
* `traj1n_frei_png`: works
* `traj2n_frei_png`: works
* `traj3n_frei_png`: works

The three cases where it failed looked like cases where the depth alone is not sufficient for camera tracking and the color information is crucial, for example here in `living_room_traj0n_frei_png`:
[image]
Unfortunately, it is expected that badslam breaks relatively easily in such situations, since its photometric residuals are not very robust.

But it should work fine in the other cases, for example for `living_room_traj1n_frei_png`:
[image]

PyojinKim, 2020-11-30:
Thank you very much for your detailed explanations and even for testing BAD SLAM on ICL-NUIM datasets directly! :)
It is totally my fault that I didn't change the order of depth and rgb words in `associated.txt` from ICL-NUIM datasets.
I get the same qualitative success / failure results when I follow your advice and setting. Thanks again!
Could I ask a few more questions if possible??

**1. BAD SLAM parameters**
I have figured out that `Maximum depth to use` in depth preprocessing tab is one of the key factors to determine the performance of BAD SLAM. In many cases, BAD SLAM works nicely when I set to 5 ~ 6 meters rather than 3 (default).
Most of the time when I set to 3, it easily jumps off and diverges because of the lack of the 3D point cloud I guess..

Btw, `Raw to metric depth conversion factor` is (1 / scale factor), right? (Just want to make sure)

**2. TAMU RGBD datasets**
I have also tested the BAD SLAM on some sequences from TAMU RGBD (http://telerobot.cs.tamu.edu/MFG/rgbd/livo/data.html).
When I run the BAD SLAM on `Corridor-A-const`, it sometimes jumps off when there are some homogeneous, non-texture walls.
(I have uploaded the above sequence on [this link](https://drive.google.com/file/d/1yI6Dhu_dWYeh8Rj7E_AghAU6ttAzluHv/view?usp=sharing) including associated.txt and calibration.txt just in case.)

[image]

It seems like the BAD SLAM sometimes fails to track the camera motion when the camera is moving forward straight ahead.
Did I understand the behavior of the BAD SLAM correctly? (BAD SLAM works well especially when there are large overlapped area.)

Thanks,
Pyojin Kim.

puzzlepaint, 2020-12-01:
1. The maximum depth setting exists mainly due to the system having targeted the first-generation Kinect in the beginning of its development. For this camera, pixels with large depth seemed so noisy that I felt it might be better to discard them completely rather than trying to use them for the algorithm. For other cameras that give more reliable depth values even if the depth is larger, it is probably good to simply disable this setting. Maybe it should thus better default to for example 6 meters instead of 3 meters.
    
    Yes, the `Raw to metric depth conversion factor` should be 1 / (the `--depth_scaling` parameter).

2. I believe that the main requirement for badslam to work reliably is that the depth image should be sufficient to constrain the camera pose. For example, if three planar walls are visible in the depth image whose normal vector directions are orthogonal, then the pose is well constrained. If only two such walls are visible in the image, then the camera pose can easily slip along the direction of the intersection of the two wall planes. This is because the photometric residuals, which would be needed to fully constrain the pose in this situation, are unfortunately quite unreliable, as mentioned above.
    
    With forward camera movement in a corridor, it is easy to obtain a situation where the depth image does not constrain the motion along the forward-backward direction sufficiently (if only the floor, ceiling, and side walls are visible), so the system is prone to such mistakes there.

PyojinKim, 2020-12-02:
I really appreciate your time and your detailed explanations and answers to my questions. :)
They help me a lot to understand the BAD SLAM algorithm. I will close this issue.

Thanks,
Pyojin Kim.

closed 2020-12-02 by PyojinKim

ZhouXiner, 2021-12-30:
Hi, I run the BAD-SLAM in ICL dataset.
I notice that if I change the trajectory values like  [tx ty tz -> tx -ty tz]  the ape can greatly improved.
                               Before change      After change 
livingRoom1              0.026682              0.003868
livingRoom2              0.170668              0.004426
office1                       0.085373              0.010144
office2                       0.185235              0.006885
office3                       0.037849              0.009318
I wonder why?  
Thanks!

puzzlepaint, 2021-12-30:
If you ran the SLAM system following the instructions above, then this is explained in my first post in this issue:
https://github.com/ETH3D/badslam/issues/59#issuecomment-735436671

The trajectories are flipped if a positive y-focal-length value is used instead of the given negative value. So they need to be flipped back before comparing them with the ground truth.