ETH3D/badslam #60: ICP-only camera tracking diverges

lanfeiying, 2020-12-04:
Hi, 
Recently I tried to implement frame-to-model camera tracking module. The registration method is ICP only. 
I implemented ICP with point-to-plane error metric, see [here](https://www.cs.princeton.edu/~smr/papers/icpstability.pdf)

I found the camera tracking module diverges after the first iteration and the error is increasing. I even tried to register two identical range image with the identity initial pose estimate. It found the correct correspondences. Theoretically, it is expected to converge and output an identity pose. 

Unfortunately, I found it diverges and outputs the wrong pose. I tested several datasets, such as TUM RGB-D dataset. It has the same problem as below: 

[image]

I am wondering if you have implemented ICP-only camera tracking module ever before? I am confused about the reason why it diverges.

puzzlepaint, 2020-12-04:
There is likely a mistake in the part of your ICP implementation that computes the relative pose based on the correspondences.

This question does not relate to badslam, does it? This issue tracker is intended for problems related to badslam only.

lanfeiying, 2020-12-04:
Hi, @puzzlepaint 
I am sorry that I put an irrelated issue on this page.
I encountered this problem when I tried to compare different front camera tracking techniques in badslam, but I cannot find out why it fails. 
I'll appreciate it if you can help me.
By the way, can I ask which computer vision community I can ask such questions? It seems that StackOverflow is not for computer vision questions.

puzzlepaint, 2020-12-07:
I am not really familiar with online Computer Vision communities. There is for example https://www.reddit.com/r/computervision/ but I don't know whether that is a suitable place for these kinds of questions. In any case, you will likely need to provide more details, otherwise one can only guess what the issue is.

closed 2020-12-07 by puzzlepaint