ETH3D/badslam #62: Some questions about the dataset in your paper

gongyeted, 2021-01-11:
1、Does the data in "groundtruth.txt" mean "global_T_frame pose" or "frame_T_global pose" ? Because in your code the output poses is "global_T_frame" poses but I consider groundtruth data in most datasets means "frame to global poses". Would you help me explicit it ?
2、Why is the rgb-d image size different than the stereo rgb2 image data size ? It may cause some troubles when using stereo images. Is it beacause some operations like undistortion or depth alignment and so on that makes rgb image size changed?
3、Trying to take advantages of imu data in your dataset to do some research. Would you provide some advice?
Thank you so much

puzzlepaint, 2021-01-13:
1. From https://www.eth3d.net/slam_documentation : "It [the ground truth pose] transforms points from the local image's frame to the global frame." So, in the A_T_B spelling convention, it should be `global_T_frame` if I'm not mistaken, such that `point_in_global == global_T_frame * point_in_frame`.
2. Yes, if I remember correctly, the final image size of the undistorted images depended on the calibration, so for a given camera, it might even differ among the datasets. As stated in the documentation, the stereo images are not stereo-rectified (the correspondences are not generally in the same image rows). Still, generic unprojection and projection operations may be used to find the point in one image that corresponds to a point in another image.
3. Personally, I have never worked with IMU data so far, so unfortunately I cannot give advice. There should be a lot of literature on this topic, however.

gongyeted, 2021-01-14:
> 1. From https://www.eth3d.net/slam_documentation : "It [the ground truth pose] transforms points from the local image's frame to the global frame." So, in the A_T_B spelling convention, it should be [...]

Thank you for your help again!