ETH3D/badslam #63: error on windows 10

ronaldpan, 2021-01-12:
When I run the windows release badslam-1.1-realsense,  
badslam-1.1-realsense>badslam D:\BadSlam\dataset\sofa_1  --gui_run --no_loop_detection
it can not run and report the error:
2021-01-12 09:48:17.970 (   0.068s) [                ]     gui_main_window.cc:1600  INFO| Read dataset with 976 frames
ARB::createContext: wglCreateContextAttribsARB() failed (GL error code: 0x502) for format: QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize 8, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile), shared context: 0x10001 
GDI::createContext: wglShareLists() failed. 
2021-01-12 09:48:36.905 (  19.003s) [                ]     kernel_opt_pose.cc:183   FATL| Cuda Error: too many resources requested for launch

Could someone help me solve the problem? Thanks.

puzzlepaint, 2021-01-13:
Not sure about the wgl error, I don't remember seeing this.

Regarding the issue `kernel_opt_pose.cc:183 FATL| Cuda Error: too many resources requested for launch`, running the auto-tuning script as described in the Readme should likely resolve that issue. Alternatively, manually create a text file `auto_tuning_result.txt` and put it into a directory named "resources" in the application executable's directory. Write the following text into the file (this is an example result of auto-tuning):
```
CalibrateDepthAndTransformColorToDepthCUDAKernel 32 16 0.0020464
UpdateSurfelPositionAndDescriptorCUDAKernel 1024 1 0.000597184
SetSurfelInactiveKernel 1024 1 0.000243616
ComputeBrightnessKernel 32 32 0.00495389
DownsampleImagesCUDAKernel 32 16 0.0135292
BilateralFilteringAndDepthCutoffCUDAKernel 8 16 0.00939517
ComputeNormalsCUDAKernel 16 16 0.00345818
DetermineSupportingSurfelsCUDAKernel 1024 1 0.00266832
CalibrateDepthCUDAKernel 32 16 0.00190826
FilterNewSurfelsCUDAKernel 1024 1 0.00015056
ComputeMinMaxPointRadiusCUDAKernel 32 16 0.00287501
AccumulatePoseEstimationCoeffsCUDAKernel 128 1 0.0372465
CompactSurfelsCUDAKernel 128 1 0.00016896
ResetSurfelAccumForSurfelDeletionAndRadiusUpdateCUDAKernel<true> 1024 1 5.2576e-05
DetermineActiveSurfelsKernel 1024 1 0.00117299
ResetSurfelForColorAssignmentKernel 1024 1 5.1104e-05
CreateSurfelsForKeyframeCUDACreationAppendKernel 8 16 0.000590368
CountObservationsAndFreeSpaceViolationsCUDAKernel<true> 1024 1 0.000849728
AccumulateColorObservationsCUDAKernel 128 1 0.00112176
CreateSurfelsForKeyframeCUDAWriteFreeSpotListKernel 128 1 9.3184e-05
ComputeCostAndResidualCountFromImagesCUDAKernel_GradientXY 8 16 0.019852
ComputeMinMaxDepthCUDAKernel 32 16 0.000247232
FlagInvalidSurfelsInAccum2CUDAKernel 256 1 0.000157344
AccumulateSurfelPositionAndDescriptorOptimizationCoeffsCUDAKernel<true> 128 1 0.0101246
UpdateSurfelNormalCUDAKernel 1024 1 0.000245408
ComputePointRadiiAndRemoveIsolatedPixelsCUDAKernel 32 16 0.00229155
ResetSurfelAccumCUDAKernel 1024 1 0.000274784
UpdateSurfelVertexBufferCUDAKernel<TEMPLATE_ARGUMENTS(false, false)> 32 1.0
AssignColorsCUDAKernel 1024 1 4.8416e-05
WriteNewSurfelIndexAndInitializeObservationsCUDAKernel 128 1 0.000325056
CountObservationsForNewSurfelsCUDAKernel 1024 1 0.00154256
ResetSurfelAccum0to3CUDAKernel 1024 1 0.000220832
AccumulatePoseEstimationCoeffsFromImagesCUDAKernel_GradientXY 8 16 0.203002
CreateSurfelsForKeyframeCUDASerializingKernel 32 32 0.00089648
AccumulateSurfelNormalOptimizationCoeffsCUDAKernel 1024 1 0.0037904
MarkDeletedSurfelsCUDAKernel 1024 1 5.7856e-05
```
If the program still fails to run, lower the block size numbers after the kernel for which it fails, using powers-of-two. For example, the failing check at `kernel_opt_pose.cc:183` is after the `AccumulatePoseEstimationCoeffsFromImages...` kernels. So, one might change the relevant line in the auto tuning file for example from:
```
AccumulatePoseEstimationCoeffsFromImagesCUDAKernel_GradientXY 8 16 0.203002
```
to:
```
AccumulatePoseEstimationCoeffsFromImagesCUDAKernel_GradientXY 8 8 0.203002
```
and retry.

ronaldpan, 2021-01-19:
OK. It works now. However, it still reported GDI::createContext: wglShareLists() failed.