ETH3D/badslam #65: question about the non keyframe frames

gongyeted, 2021-03-04:
In tracking thread you track every new frame by align to the closest keyframe to get poses of every frame, but all surfels are created by keyframes in BA thread. What does the poses of non key frames do? It seems that they don't make any contributions to the surfels showed in the system or even to the keyframes' poses. Is it right?

puzzlepaint, 2021-03-08:
The poses of the non-keyframes are used in the prediction of the following frames' poses. So, they do help to make the pose estimation converge with higher probability, by providing better initial pose estimates for those frames.

gongyeted, 2021-03-10:
> The poses of the non-keyframes are used in the prediction of the following frames' poses. So, they do help to make the pose estimation converge with higher probability, by providing better initial pose estimates for those frames.

Thank you for your reply.