ETH3D/badslam #71: DLib_LIBRARY NOTFOUND

dianaleeg, 2021-08-12:
I have been having issues with getting cmake to recognize **DLib_LIBRARY**. I have DLib in **/usr/local/lib** and I see that my **DlibConfig.cmake** file looks like this:
```
FIND_LIBRARY(DLib_LIBRARY DLib
    PATHS "/usr/local/lib"
)
FIND_PATH(DLib_INCLUDE_DIR DLibConfig.cmake
    PATHS "/usr/local/lib/DLib/build" 
)
LIST(APPEND DLib_INCLUDE_DIR
  ${DLib_INCLUDE_DIR}/../ ${DLib_INCLUDE_DIR}/../DUtils
  ${DLib_INCLUDE_DIR}/../DUtilsCV ${DLib_INCLUDE_DIR}/../DVision)
SET(DLib_LIBRARIES ${DLib_LIBRARY})
SET(DLib_LIBS ${DLib_LIBRARY})
SET(DLib_INCLUDE_DIRS ${DLib_INCLUDE_DIR})
```

I was able to get rid of an error when i updated DLib_INCLUDE_DIR to point to my /build area but I can't for the life of me figure out what's going on with DLib_LIBRARY and i tried to point it to various locations within DLib.

**Here is my error:** 
```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
DLib_LIBRARY
    linked by target "badslam_baselib" in directory /home/dianaleeg/code/badslam/applications/badslam
    linked by target "DBoW2" in directory /home/dianaleeg/code/badslam/applications/badslam/third_party/DBoW2
    linked by target "demo" in directory /home/dianaleeg/code/badslam/applications/badslam/third_party/DBoW2
```

Can someone possibly point me to what I'm clearly not seeing?

puzzlepaint, 2021-08-12:
I'm afraid I can't point to anything specific. In general, if you had to modify the DlibConfig.cmake file, it sounds to me like something may be wrong with the DLib installation, since I would normally expect that to work out-of-the-box.

On my current system, `DLib_LIBRARY` is set to `<dlib's_installation_directory>/lib/libDLib.so`. So, I guess that looking for the `libDLib.so` file and pointing the search path in DlibConfig.cmake to its directory may be a workaround.

Also notice that there are at least two libraries called "DLib", badslam requires [this one](https://github.com/dorian3D/dlib). If the DlibConfig.cmake file you quoted was installed by the library, then it seems to be the correct one though.

dianaleeg, 2021-08-13:
Hey there @puzzlepaint , i appreciate the response! Hmm, im wondering if it has to do with the way I am using cmake? I created a new directory 'build' inside of DLib, then i navigate to the folder 'build' and then run cmake .. . 

($ mkdir build, $ cd build, $ cmake .. ) 

I did notice that I dont have a shared object file -- is this supposed to be a result of cmake? I don't see it in the repository either.

(I can verify that that is the DLib that I installed!)

dianaleeg, 2021-08-13:
AH-HA, i was missing $ make 

Thank you very kindly for your response. All is working!

closed 2021-08-13 by dianaleeg

puzzlepaint, 2021-08-13:
Usually, one would also run `make install` after `make`, possibly with `sudo` if necessary. This copies all the relevant output files produced by `make` to the installation location (which is likely under the /usr/local prefix by default), where they are supposed to be found later by CMake for badslam (if CMake is told to use this installation rather than the path you used before).

Perhaps omitting the installation causes the include files directory not being set properly as in #37.

dianaleeg, 2021-08-14:
AH yes! That did solve my issue, thank you very much for your help.