ETH3D/badslam #73: Symbol lookup error

mshong0320, 2021-08-31:
I've successfully built badslam from source but when I try to run the binaries (build/application/badslam/badslam) I get a symbol lookup error .
/badslam: undefined symbol: _ZN5DBoW26FBrief8toMat32FERKSt6vectorIN5boost14dynamic_bitsetImSaImEEESaIS5_EERN2cv3MatE

I have a feeling this has to do with my boost version I am using (1.65) but if you think it has to do with something else please do let me know.

This is the output from my cmake cmd:
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for a CUDA compiler
-- Looking for a CUDA compiler - /usr/local/cuda/bin/nvcc
-- The CUDA compiler identification is NVIDIA 10.2.89
-- Check for working CUDA compiler: /usr/local/cuda/bin/nvcc
-- Check for working CUDA compiler: /usr/local/cuda/bin/nvcc -- works
-- Detecting CUDA compiler ABI info
-- Detecting CUDA compiler ABI info - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "2.91.0") 
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   serialization
--   filesystem
--   system
-- Found PythonInterp: /usr/bin/python (found version "2.7.17") 
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
-- Performing Test HAVE_LD_VERSION_SCRIPT
-- Performing Test HAVE_LD_VERSION_SCRIPT - Success
-- Symbol prefix: 
-- Found GLEW: /usr/include  
-- Found CUDA: /usr/local/cuda (found suitable exact version "10.2") 
-- Found OpenCV: /usr/local/opencv346 (found suitable version "3.4.6", minimum required is "3.4.6") 
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Found OpenCV: /usr/local/opencv346 (found version "3.4.6") 
-- Boost version: 1.65.1
DLib library found, using it from the system
CMake Warning at applications/badslam/CMakeLists.txt:34 (find_package):
  By not providing "Findk4a.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "k4a", but
  CMake did not find one.

  Could not find a package configuration file provided by "k4a" with any of
  the following names:

    k4aConfig.cmake
    k4a-config.cmake

  Add the installation prefix of "k4a" to CMAKE_PREFIX_PATH or set "k4a_DIR"
  to a directory containing one of the above files.  If "k4a" provides a
  separate development package or SDK, be sure it has been installed.

CMake Warning at applications/badslam/CMakeLists.txt:35 (find_package):
  By not providing "Findk4arecord.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "k4arecord", but CMake did not find one.

  Could not find a package configuration file provided by "k4arecord" with
  any of the following names:

    k4arecordConfig.cmake
    k4arecord-config.cmake

  Add the installation prefix of "k4arecord" to CMAKE_PREFIX_PATH or set
  "k4arecord_DIR" to a directory containing one of the above files.  If
  "k4arecord" provides a separate development package or SDK, be sure it has
  been installed.

-- SCSDK_ROOT has not been set. If you would like to use the Structure SDK support, please set this to the root path of the Structure SDK manually.
-- The following OPTIONAL packages have been found:

 * PythonInterp
 * realsense2

-- The following REQUIRED packages have been found:

 * ZLIB
 * GLEW
 * Qt5Core
 * Qt5Gui (required version >= 5.9.5)
 * Qt5OpenGL
 * Qt5Widgets
 * Qt5 (required version >= 5.8)
 * Qt5X11Extras
 * CUDA (required version == 10.2)
 * SuiteSparse
 * Eigen3
 * g2o
 * OpenGL
 * Threads
 * X11
 * OpenCV
 * Boost
 * DLib
 * opengv

-- The following OPTIONAL packages have not been found:

 * k4a
 * k4arecord
 * Structure

-- Configuring done
-- Generating done
-- Build files have been written to: /home/minsung/badslam/build

puzzlepaint, 2021-09-02:
I think that this happens if g2o and badslam are compiled with different, incompatible compiler options.
Quoting from the Readme:

> If the Optimization.PoseGraphOptimizer test crashes, for example with an error message like "Cholesky failure", then please verify that your build of g2o has the BUILD_WITH_MARCH_NATIVE CMake option set to ON, and that BAD SLAM actually uses this install of g2o.

mshong0320, 2021-09-02:
Got it thank you!