ETH3D/badslam #74: External odometry source implementation

mshong0320, 2021-09-02:
Hi,

I'd like to try to use an external odometry source instead of the visual odometry node updating the pose.

Could you guide me where and how i should make the change on this repo?

Thank you in advance

puzzlepaint, 2021-09-04:
You could replace the implementation of the `RunOdometry()` function in `bad_slam.cc`:
https://github.com/ETH3D/badslam/blob/32e47bee6d4cdac81c5603c6243eef55adb2f719/applications/badslam/src/badslam/bad_slam.cc#L827

This function returns the estimated pose in the lines starting from here, where you can set in your own pose instead:
https://github.com/ETH3D/badslam/blob/32e47bee6d4cdac81c5603c6243eef55adb2f719/applications/badslam/src/badslam/bad_slam.cc#L937

mshong0320, 2021-09-09:
I'm trying to use realsense T265 tracking camera with a D435. I'm not sure how to instantiate an object of a T265 camera on the BadSlam::Badslam() of badslam.cc file. How does D435 get instantiated? Could you please tell me how I can do this?

puzzlepaint, 2021-09-09:
This question is beyond the scope of this issue tracker for the badslam application. Unfortunately, I don't have the time to provide general programming support.

closed 2021-09-09 by puzzlepaint