ETH3D/badslam #75: badslam-test passed 11 test but failed 3 tests in a Jetson AGX Xavier

KyleXBao, 2021-09-27:
System: Jetson AGX Xavier
OS: Ubuntu 18.04
Jetpack: 4.5.1
Cuda: 10.2
Opencv: build from source 3.4.15 with CUDA
Boosts:  1.65.0
DLib:	commit b6c28fb with CUDA
Eigen:	3.3.7
g2o	with cmake -DBUILD_WITH_MARCH_NATIVE=ON ..
GLEW	
GTest	
OpenGV
Qt	5.12.0
badslam: cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_CUDA_FLAGS="-arch=sm_72" ..

I firstly downloaded brief_k10L6.voc and brief_pattern.yml and installed in folder: build/applications/badslam/resources .
Then, I installed dataset/cables_3 and run python scripts/auto_tune_parameters.py . After that, I copied the generated auto_tuning_result.txt it to folder: build/applications/badslam/resources
Before I start badslam, I run the test file badslam-test. The results has 10 passed and 4 failed. The logs are as follows:

2mtime                            file:line     v| 
2m14:03:52.769              loguru.cpp:587   INFO| arguments: ./badslam_test
2m14:03:52.769              loguru.cpp:590   INFO| Current dir: /home/robot/badslam/build/applications/badslam
2m14:03:52.769              loguru.cpp:592   INFO| stderr verbosity: 2
2m14:03:52.769              loguru.cpp:593   INFO| -----------------------------------
[==========] Running 14 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 14 tests from Optimization
[ RUN      ] Optimization.AlternatingGeometryOptimizationWithGeometricResidual
[       OK ] Optimization.AlternatingGeometryOptimizationWithGeometricResidual (529 ms)
[ RUN      ] Optimization.PCGGeometryOptimizationWithGeometricResidual
2m14:03:53.575        direct_ba_pcg.cc:748   INFO| Debug: early convergence in PCG-based BA at iteration: 9
2m14:03:53.638        direct_ba_pcg.cc:748   INFO| Debug: early convergence in PCG-based BA at iteration: 9
2m14:03:53.702        direct_ba_pcg.cc:748   INFO| Debug: early convergence in PCG-based BA at iteration: 9
2m14:03:53.766        direct_ba_pcg.cc:748   INFO| Debug: early convergence in PCG-based BA at iteration: 9
2m14:03:53.830        direct_ba_pcg.cc:748   INFO| Debug: early convergence in PCG-based BA at iteration: 9
2m14:03:53.895        direct_ba_pcg.cc:748   INFO| Debug: early convergence in PCG-based BA at iteration: 9
2m14:03:53.961        direct_ba_pcg.cc:748   INFO| Debug: early convergence in PCG-based BA at iteration: 9
2m14:03:54.027        direct_ba_pcg.cc:748   INFO| Debug: early convergence in PCG-based BA at iteration: 9
2m14:03:54.091        direct_ba_pcg.cc:748   INFO| Debug: early convergence in PCG-based BA at iteration: 9
2m14:03:54.156        direct_ba_pcg.cc:748   INFO| Debug: early convergence in PCG-based BA at iteration: 9
2m14:03:54.157        direct_ba_pcg.cc:773   INFO| Timing
------
BA surfel activation                	     100	555272395205654307611881766912.0000	(5552723952056542944177422336.000000 +- 4556620468368619302748684288.000000)	[00.000125,9254539920094238973303455744.000000]
BA geometry optimization            	     100	555272395205654307611881766912.0000	(5552723952056542944177422336.000000 +- 4556620468368619302748684288.000000)	[00.000125,9254539920094238973303455744.000000]
BA PCG step                         	     100	000000.5920	(00.005920 +- 00.000152)	[00.005688,00.007055]
BA normals update                   	     100	000000.0305	(00.000305 +- 00.000007)	[00.000288,00.000328]
BA final surfel del. and radius upd.	      20	000000.0082	(00.000410 +- 00.000219)	[00.000292,00.001075]

[       OK ] Optimization.PCGGeometryOptimizationWithGeometricResidual (717 ms)
[ RUN      ] Optimization.AlternatingGeometryOptimizationWithPhotometricResidual
2m14:03:54.387 test_geometry_optimizat:105   INFO| num_fails: 33999 (13.4803% of surfels; surfel_count: 252212)
[       OK ] Optimization.AlternatingGeometryOptimizationWithPhotometricResidual (192 ms)
[ RUN      ] Optimization.PCGGeometryOptimizationWithPhotometricResidual
2m14:03:56.698        direct_ba_pcg.cc:748   INFO| Debug: early convergence in PCG-based BA at iteration: 59
2m14:03:56.729 test_geometry_optimizat:105   INFO| num_fails: 34586 (13.6826% of surfels; surfel_count: 252774)
[       OK ] Optimization.PCGGeometryOptimizationWithPhotometricResidual (2344 ms)
[ RUN      ] Optimization.AlternatingDepthDeformationOptimizationWithGeometricResidual
/home/robot/badslam/applications/badslam/src/badslam/test/test_intrinsics_optimization_geometric_residual.cc:347: Failure
The difference between true_a and direct_ba.a() is 0.04270579107105732, which exceeds 1e-2f, where
true_a evaluates to 0.029999999329447746,
direct_ba.a() evaluates to 0.072705790400505066, and
1e-2f evaluates to 0.0099999997764825821.
[  FAILED  ] Optimization.AlternatingDepthDeformationOptimizationWithGeometricResidual (29598 ms)
[ RUN      ] Optimization.PCGDepthDeformationOptimizationWithGeometricResidual
2m14:04:31.660        direct_ba_pcg.cc:748   INFO| Debug: early convergence in PCG-based BA at iteration: 0
2m14:04:31.757        direct_ba_pcg.cc:748   INFO| Debug: early convergence in PCG-based BA at iteration: 0
2m14:04:31.922        direct_ba_pcg.cc:748   INFO| Debug: early convergence in PCG-based BA at iteration: 0
2m14:04:32.031        direct_ba_pcg.cc:748   INFO| Debug: early convergence in PCG-based BA at iteration: 0
2m14:04:32.143        direct_ba_pcg.cc:748   INFO| Debug: early convergence in PCG-based BA at iteration: 0
2m14:04:32.257        direct_ba_pcg.cc:748   INFO| Debug: early convergence in PCG-based BA at iteration: 0
[... 513 lines cut ...]
2m14:05:52.502 test_pairwise_frame_tra:543   INFO| Distortion strenght: 0.05 --> average error norm (convergence): 0.00111627 --> average error norm (accuracy): 0.000687628
2m14:05:52.710 test_pairwise_frame_tra:543   INFO| Distortion strenght: 0.025 --> average error norm (convergence): 0.00112235 --> average error norm (accuracy): 0.000536803
2m14:05:52.899 test_pairwise_frame_tra:543   INFO| Distortion strenght: 0.01 --> average error norm (convergence): 0.00112283 --> average error norm (accuracy): 0.000340571
2m14:05:53.072 test_pairwise_frame_tra:543   INFO| Distortion strenght: 0.001 --> average error norm (convergence): 0.00111607 --> average error norm (accuracy): 0.000932091
[       OK ] Optimization.PairwiseFrameTracking (4481 ms)
[ RUN      ] Optimization.PoseGraphOptimizer
2m14:05:53.093 pose_graph_optimizer.cc:127   INFO| - Performing pose graph optimization ...
2m14:05:53.096 pose_graph_optimizer.cc:131   INFO| - Pose graph optimization done
[       OK ] Optimization.PoseGraphOptimizer (21 ms)
[ RUN      ] Optimization.PoseOptimizationWithGeometricResidual
[       OK ] Optimization.PoseOptimizationWithGeometricResidual (22 ms)
[ RUN      ] Optimization.PoseOptimizationColorOnlyCues
[       OK ] Optimization.PoseOptimizationColorOnlyCues (36 ms)
[----------] 14 tests from Optimization (120207 ms total)

[----------] Global test environment tear-down
[==========] 14 tests from 1 test case ran. (120207 ms total)
[  PASSED  ] 11 tests.
[  FAILED  ] 3 tests, listed below:
[  FAILED  ] Optimization.AlternatingDepthDeformationOptimizationWithGeometricResidual
[  FAILED  ] Optimization.AlternatingIntrinsicsOptimizationWithPhotometricResidual
[  FAILED  ] Optimization.PCGIntrinsicsOptimizationWithPhotometricResidual

 3 FAILED TESTS
2m14:05:53.167              loguru.cpp:467   INFO| atexit

puzzlepaint, 2021-09-28:
Thank you for reporting these failures. I believe that they are likely more of an issue with the tests rather than an issue with the tested functionality. There is generally some variance in the test results, for example due to differing CUDA block scheduling. It seems that the thresholds used in the tests may be slightly too small to account for that variance.

In detail:

The test `Optimization.PCGIntrinsicsOptimizationWithPhotometricResidual` reached a difference of `-0.0406952, 0.0275574, -0.0578613, -0.114563` to the ground truth intrinsics, but the thresholds on the absolute difference values are `0.03, 0.03, 0.15, 0.15` (only the first value exceeds the threshold slightly).

The test `Optimization.AlternatingIntrinsicsOptimizationWithPhotometricResidual` reached a difference of `-0.0420685, 0.027298, -0.0567017, -0.11319` to the ground truth intrinsics, but the thresholds on the absolute difference values are `0.03, 0.03, 0.15, 0.15` (again, only the first value exceeds the threshold slightly).

The test `Optimization.AlternatingDepthDeformationOptimizationWithGeometricResidual` exceeded a threshold of `0.01` more significantly with a value of `0.0427`, but there is a comment in the test code that this specific value converges very slowly and probably does not matter much.

KyleXBao, 2021-09-28:
Thank you for your explanation. I was worried about these 3 to 4 test failures because it's written all tests should passed in your installation instruction. Since it's not matter much, I can continue work on the badslam.