ETH3D/badslam #79: Target "badslam_test" links to target "Eigen3::Eigen" but the target was   not found.

lql0602, 2021-11-17:
Thanks for your work. However, I cannot build it now. I just input: cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_CUDA_FLAGS="-arch=sm_61" .. 

CMake Error at applications/badslam/CMakeLists.txt:200 (add_executable):
  Target "badslam_test" links to target "Eigen3::Eigen" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at applications/badslam/CMakeLists.txt:200 (add_executable):
  Target "badslam_test" links to target "Eigen3::Eigen" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at applications/badslam/CMakeLists.txt:200 (add_executable):
  Target "badslam_test" links to target "Eigen3::Eigen" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at applications/badslam/CMakeLists.txt:200 (add_executable):
  Target "badslam_test" links to target "Eigen3::Eigen" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at applications/badslam/CMakeLists.txt:180 (add_executable):
  Target "badslam" links to target "Eigen3::Eigen" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at applications/badslam/CMakeLists.txt:180 (add_executable):
  Target "badslam" links to target "Eigen3::Eigen" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at applications/badslam/CMakeLists.txt:180 (add_executable):
  Target "badslam" links to target "Eigen3::Eigen" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at applications/badslam/CMakeLists.txt:180 (add_executable):
  Target "badslam" links to target "Eigen3::Eigen" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at applications/badslam/CMakeLists.txt:56 (add_library):
  Target "badslam_baselib" links to target "Eigen3::Eigen" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at applications/badslam/CMakeLists.txt:56 (add_library):
  Target "badslam_baselib" links to target "Eigen3::Eigen" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at applications/badslam/CMakeLists.txt:56 (add_library):
  Target "badslam_baselib" links to target "Eigen3::Eigen" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at applications/badslam/CMakeLists.txt:56 (add_library):
  Target "badslam_baselib" links to target "Eigen3::Eigen" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

but i have instaledl eigen3
-modversion eigen3
3.3.7

puzzlepaint, 2021-11-17:
This seems like there must be some error in locating the Eigen3 installation. Badslam's CMakeLists.txt does have a `find_package` call for it here:
https://github.com/ETH3D/badslam/blob/32e47bee6d4cdac81c5603c6243eef55adb2f719/CMakeLists.txt#L72

Perhaps Eigen3 is installed in a non-standard location on your system? In that case, you could try specifying its path to CMake manually (for example, by checking with `ccmake` and filling in the Eigen3-related variables that are set to NOT_FOUND)

neilzhuu, 2021-11-22:
Hello @lql0602, I got the same problem. Did you solve it?

neilzhuu, 2021-11-22:
I replace `find_package(Eigen3 REQUIRED) ` with `find_package(Eigen3 REQUIRED NO_MODULE) ` and  finally it compiles well.

puzzlepaint, 2021-11-23:
@neilzhuu Thanks for the update. Eigen also seems to recommend using NO_MODULE (see [Using Eigen in CMake Projects](https://eigen.tuxfamily.org/dox/TopicCMakeGuide.html)), so I [committed](https://github.com/ETH3D/badslam/commit/e7ce8177adf53a47486a60213f774795b2b38754) the change to the repository.

closed 2021-11-23 by puzzlepaint