ETH3D/badslam #80: About BADSLAM2 Results on ETH3D SLAM Benchmark

II-Matto, 2021-11-23:
I noticed there are recently submitted results to ETH3D SLAM benchmark, which are named as BADSLAM2. Are those results reproducible with the code in this repository? Or is BADSLAM2 a new work? If so, is there a paper available? Many thanks.

puzzlepaint, 2021-11-23:
1. The results of the code in this repository here are supposed to differ slightly from the "BAD SLAM" benchmark results since the open source version is slightly different from the initial version used in the paper, as mentioned in the "Differences to the paper" section at the end of the [Readme](https://github.com/ETH3D/badslam/blob/master/README.md). However, someone anonymously submitted a result "[badslam_opensource](https://www.eth3d.net/result_details?id=232)" to the benchmark which shows a result obtained with the open source version. In the [overall comparison](https://www.eth3d.net/slam_benchmark), it can be seen that the two results are very similar on the test set, with the open source version appearing to give slightly better accuracy, but slightly worse robustness, compared to the paper version. The anonymous submission does unfortunately not include results for the training set, but one could expect the results to be similar there as well.
2. This is a new work, and I am not involved in it. So far I am not aware of a paper for it. I would be interested in reading about it as well :)

II-Matto, 2021-11-25:
Thanks for your detailed explanation!

closed 2021-11-25 by II-Matto