ETH3D/badslam #85: Build with CUDA 11.6

irenji, 2022-04-22:
Currently I'm trying to compile with CUDA 11.6, but badslam compline about CUB dependency. Seems that in CUDA 11 it's try to use version of CUB that is installed by CUDA.
Is there any test with CUDA 11.x?

puzzlepaint, 2022-04-24:
Since the following commit, the project uses the version of CUB that is distributed with CUDA:
https://github.com/ETH3D/badslam/commit/1fb449eb3f31db98cdbdc9f19e01c588c9e99f9d

Also, this commit fixes a bug caused by a breaking change in one of the newer versions of CUB:
https://github.com/ETH3D/badslam/commit/e5cb94b38671015aeb5c3faa45a52a8cb24159e9

After that second commit, it definitely worked with CUDA 11.x. What is the error that you encountered?

irenji, 2022-04-24:
This is the error, It doesn't depend on architecture, 

[image]

`1>Compiling CUDA source file ..\..\..\applications\badslam\src\badslam\cuda_image_processing.cu...
1>Compiling CUDA source file ..\..\..\applications\badslam\src\badslam\cuda_depth_processing.cu...
1>
1>D:\badslam\badslam\build\applications\badslam>"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.6\bin\nvcc.exe" -gencode=arch=compute_52,code=\"sm_52,compute_52\" --use-local-env -ccbin "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX64\x64" -x cu   -ID:\badslam\badslam\build\applications\badslam -ID:\badslam\badslam\applications\badslam -ID:\badslam\badslam\build\applications\badslam\badslam_baselib_autogen\include_Release -ID:\badslam\badslam\applications\badslam\src -ID:\badslam\badslam\applications\badslam\third_party\DBoW2\include -I"D:\3rdparty\vcpkg\installed\x64-windows\include\suitesparse" -ID:\badslam\badslam\applications\badslam\third_party -I"D:\3rdparty\vcpkg\installed\x64-windows\include\eigen3" -ID:\badslam\badslam\libvis\third_party\sophus -ID:\badslam\badslam\.\libvis\src -ID:\badslam\badslam\libvis\third_party\loguru -I"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.6\include" -I"D:\3rdparty\vcpkg\installed\x64-windows\include" -ID:\badslam\badslam\applications\badslam\third_party\opengv\include -I"D:\3rdparty\vcpkg\installed\x64-windows\share\g2o\..\..\include" -I"D:\3rdparty\vcpkg\installed\x64-windows\include\qt5" -I"D:\3rdparty\vcpkg\installed\x64-windows\include\qt5\QtWidgets" -I"D:\3rdparty\vcpkg\installed\x64-windows\include\qt5\QtGui" -I"D:\3rdparty\vcpkg\installed\x64-windows\include\qt5\QtCore" -I"D:\3rdparty\vcpkg\installed\x64-windows\tools\qt5\mkspecs\win32-msvc" -I"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.6\include"     --keep-dir x64\Release -use_fast_math -maxrregcount=0  --machine 64 --compile -cudart static --expt-relaxed-constexpr -std=c++14 -Xcompiler="/EHsc -Ob2"   -D_WINDOWS -DNDEBUG -DLIBVIS_HAVE_QT -DLIBVIS_HAVE_OPENCV -DLIBVIS_HAVE_CUDA -DNOMINMAX -DNPARTITION -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_NO_DEBUG -D"CMAKE_INTDIR=\"Release\"" -D_MBCS -DWIN32 -D_WINDOWS -DNDEBUG -DLIBVIS_HAVE_QT -DLIBVIS_HAVE_OPENCV -DLIBVIS_HAVE_CUDA -DNOMINMAX -DNPARTITION -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_NO_DEBUG -D"CMAKE_INTDIR=\"Release\"" -Xcompiler "/EHsc /W3 /nologo /O2 /FdD:\badslam\badslam\build\applications\badslam\Release\badslam_baselib.pdb /FS   /MD /GR" -o badslam_baselib.dir\Release\/src/badslam/cuda_image_processing.cu.obj "D:\badslam\badslam\applications\badslam\src\badslam\cuda_image_processing.cu"
1>
1>D:\badslam\badslam\build\applications\badslam>"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.6\bin\nvcc.exe" -gencode=arch=compute_52,code=\"sm_52,compute_52\" --use-local-env -ccbin "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX64\x64" -x cu   -ID:\badslam\badslam\build\applications\badslam -ID:\badslam\badslam\applications\badslam -ID:\badslam\badslam\build\applications\badslam\badslam_baselib_autogen\include_Release -ID:\badslam\badslam\applications\badslam\src -ID:\badslam\badslam\applications\badslam\third_party\DBoW2\include -I"D:\3rdparty\vcpkg\installed\x64-windows\include\suitesparse" -ID:\badslam\badslam\applications\badslam\third_party -I"D:\3rdparty\vcpkg\installed\x64-windows\include\eigen3" -ID:\badslam\badslam\libvis\third_party\sophus -ID:\badslam\badslam\.\libvis\src -ID:\badslam\badslam\libvis\third_party\loguru -I"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.6\include" -I"D:\3rdparty\vcpkg\installed\x64-windows\include" -ID:\badslam\badslam\applications\badslam\third_party\opengv\include -I"D:\3rdparty\vcpkg\installed\x64-windows\share\g2o\..\..\include" -I"D:\3rdparty\vcpkg\installed\x64-windows\include\qt5" -I"D:\3rdparty\vcpkg\installed\x64-windows\include\qt5\QtWidgets" -I"D:\3rdparty\vcpkg\installed\x64-windows\include\qt5\QtGui" -I"D:\3rdparty\vcpkg\installed\x64-windows\include\qt5\QtCore" -I"D:\3rdparty\vcpkg\installed\x64-windows\tools\qt5\mkspecs\win32-msvc" -I"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.6\include"     --keep-dir x64\Release -use_fast_math -maxrregcount=0  --machine 64 --compile -cudart static --expt-relaxed-constexpr -std=c++14 -Xcompiler="/EHsc -Ob2"   -D_WINDOWS -DNDEBUG -DLIBVIS_HAVE_QT -DLIBVIS_HAVE_OPENCV -DLIBVIS_HAVE_CUDA -DNOMINMAX -DNPARTITION -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_NO_DEBUG -D"CMAKE_INTDIR=\"Release\"" -D_MBCS -DWIN32 -D_WINDOWS -DNDEBUG -DLIBVIS_HAVE_QT -DLIBVIS_HAVE_OPENCV -DLIBVIS_HAVE_CUDA -DNOMINMAX -DNPARTITION -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_NO_DEBUG -D"CMAKE_INTDIR=\"Release\"" -Xcompiler "/EHsc /W3 /nologo /O2 /FdD:\badslam\badslam\build\applications\badslam\Release\badslam_baselib.pdb /FS   /MD /GR" -o badslam_baselib.dir\Release\/src/badslam/cuda_depth_processing.cu.obj "D:\badslam\badslam\applications\badslam\src\badslam\cuda_depth_processing.cu"
1>C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v11.6/include\cub/device/dispatch/dispatch_segmented_sort.cuh(338): error : invalid combination of type specifiers
1>
1>C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v11.6/include\cub/device/dispatch/dispatch_segmented_sort.cuh(338): error : expected an identifier
1>
[... cut ...]  -o badslam_baselib.dir\Release\/src/badslam/cuda_depth_processing.cu.obj "D:\badslam\badslam\applications\badslam\src\badslam\cuda_depth_processing.cu"" exited with code 2.
1>Done building project "badslam_baselib.vcxproj" -- FAILED.
2>------ Build started: Project: badslam (badslam\badslam), Configuration: Release x64 ------
2>Automatic MOC for target badslam
2>LINK : fatal error LNK1181: cannot open input file 'Release\badslam_baselib.lib'
2>Done building project "badslam.vcxproj" -- FAILED.`

puzzlepaint, 2022-04-25:
I found this issue reported for CUB where exactly the same error messages were reported:
https://github.com/NVIDIA/cub/issues/422

So it seems that it is a Windows-specific problem. I don't have time now to test this on Windows, but it should be simple to fix by ensuring that Windows.h is never included before including CUB.

irenji, 2022-04-27:
I had been able to fix it with minimal change #86

closed 2022-04-27 by puzzlepaint