ETH3D/badslam #87: FATL| RealSense input requested, but the program was compiled without RealSense support.

AreteQin, 2022-05-07:
Hello,

I met a problem while I was trying to run badslam with arguments "live://realsense". I am trying to start badslam without gui using realsense, as shown below.

```C++
/home/qin/Downloads/badslam_without_loop_dector/cmake-build-debug/applications/badslam/badslam live://realsense
time                            file:line     v| 
20:33:44.731              loguru.cpp:587   INFO| arguments: /home/qin/Downloads/badslam_without_loop_dector/cmake-build-debug/applications/badslam/badslam live://realsense
20:33:44.731              loguru.cpp:590   INFO| Current dir: /home/qin/Downloads/badslam_without_loop_dector/cmake-build-debug/applications/badslam
20:33:44.731              loguru.cpp:592   INFO| stderr verbosity: 2
20:33:44.731              loguru.cpp:593   INFO| -----------------------------------
Stack trace:
4       0x55773e57590e /home/qin/Downloads/badslam_without_loop_dector/cmake-build-debug/applications/badslam/badslam(+0x4590e) [0x55773e57590e]
3       0x7f7dda1c50b3 __libc_start_main + 243
2       0x55773e575156 /home/qin/Downloads/badslam_without_loop_dector/cmake-build-debug/applications/badslam/badslam(+0x45156) [0x55773e575156]
1       0x55773e57953f /home/qin/Downloads/badslam_without_loop_dector/cmake-build-debug/applications/badslam/badslam(+0x4953f) [0x55773e57953f]
0       0x7f7ddb96b49f loguru::StreamLogger::~StreamLogger() + 113
20:33:44.828       input_realsense.h:99    FATL| RealSense input requested, but the program was compiled without RealSense support.

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)

```

But if I run badslam with gui, it is working properly. Please forgive me if I missed anything already updated.
Thank you for your help in advance.

puzzlepaint, 2022-05-09:
I don't see how this could happen. Looking at the following file:
https://github.com/ETH3D/badslam/blob/master/applications/badslam/src/badslam/input_realsense.h
It seems that either the RealSense support gets compiled into the binary, or the fatal error message at `input_realsense.h:99`, but never both at the same time. Thus, I suspect that when you ran badslam with gui, it was a different binary than the one that generated the log quoted above. At least I don't see any other possibility at the moment.

AreteQin, 2022-05-11:
Thanks for your response.

I double-checked, and I can confirm that I am running the same binary file.

puzzlepaint, 2022-05-11:
Thanks for double-checking. I think I found the issue: The define `BAD_SLAM_HAVE_REALSENSE` is mistakenly added only to the `badslam_baselib` target, rather than also to the executable target. That's why the [input_realsense.h](https://github.com/ETH3D/badslam/blob/master/applications/badslam/src/badslam/input_realsense.h) header is compiled differently when included from the latter target.

In `applications/badslam/CMakeLists.txt`:
```cmake
    target_compile_options(badslam_baselib
      PRIVATE
        $<$<COMPILE_LANGUAGE:CXX>:-DBAD_SLAM_HAVE_REALSENSE>
    )
```
While I can't test it right now, I believe that changing the "PRIVATE" to "PUBLIC" here should be the proper fix for this issue.
Sorry for suspecting a wrong cause.

AreteQin, 2022-05-11:
Thanks a lot. I changed private to public, and it's working well now.

closed 2022-05-11 by AreteQin