ETH3D/badslam #89: angle_difference in Direct BA

AreteQin, 2022-05-15:
Hi,
While I am looking at the code of the function, void DirectBA::MergeKeyframes(), in [Direct BA](https://raw.githubusercontent.com/ETH3D/badslam/master/applications/badslam/src/badslam/direct_ba.cc), I found that when you are calculating the angle_difference, you are using only the third column of rotationMatrix(), as shown below:
```C++
float angle_difference = acosf(keyframe->global_T_frame().rotationMatrix().block<3, 1>(0, 2).dot(next_keyframe->global_T_frame().rotationMatrix().block<3, 1>(0, 2)));
```

But I don't understand why are you only using the third column if you are trying to calculate the angle difference of three directions, since as I know the angle difference is calculated as below. Or do you have other considerations?
$$
2 \cos (|\alpha|)=\operatorname{trace}\left(\mathrm{R}_{1}^{-1} \mathrm{R}_{2}\right)-1
$$

Please forgive me if I missed anything already updated.
Thank you for your help in advance.

puzzlepaint, 2022-05-16:
The goal of the MergeKeyframes() function is to find keyframes that observe almost the same view of the scene, which means that they are mostly redundant and can be merged / deleted without causing much trouble. This criterion is approximated with a heuristical difference metric, which is based on the positional distance and the version of an angle distance that you quoted.

The idea behind using only the third column was that this should correspond to the "forward" direction of the keyframes, which is the main part of the direction that determines where a keyframe is looking at. Two hypothetical keyframes that have equal position and have the same "forward" direction, but differ in the remaining axes of the orientation, still mostly observe the same part of the scene because they look in the same direction and only differ by some "roll" rotation (a rotation around the "forward" direction) - at least assuming that the camera frustum has similar extents on its sides. Thus, the heuristic only takes that part of the orientation difference into account.

Of course, all of this is only a heuristic and two frames that are very close to each other in position and orientation could still observe strongly different parts of the scene, for example, if they are on different sides of some kind of obstacle. However, a more elaborate criterion (for example, based on the depth images) would be slower to compute.

AreteQin, 2022-05-16:
I understood. Thank you so much for your exhaustive explanation.

closed 2022-05-16 by AreteQin