ETH3D/badslam #92: Let libpng expand the color channels to 16 bit if requested

Starry-lei, 2022-12-29:
I guess it is a bug that the code for "Let libpng expand the color channels to 16 bit if requested"  is written as " if (output_bit_depth == 16 && bit_depth < 16)".   (libvis/src/libvis/image_io_libpng.cc)

it should be  if (output_bit_depth == 16 && bit_depth ==16), right :) ?

puzzlepaint, 2023-01-01:
Why do you think that it should be changed to that?

The intention here is that if the output image has 16 bits per pixel, the code asks libpng to scale the PNG image data to 16 bits per pixel as well, by calling `png_set_expand_16()`. If the output image and the PNG data already both have 16 bits per pixel, then no action is necessary.

Starry-lei, 2023-01-01:
Thanks for the reply, now I got it, you are right.

closed 2023-01-01 by Starry-lei