ETH3D/badslam #93: Can we use 16-bit RGB image as input without converting into 8-bit RGB(by libpng)?

Starry-lei, 2023-01-01:
Hi, I think the code  here:
" image->SetSize(width, height);
  png_bytep* row_pointers = new png_bytep[height];
  for (u32 y = 0; y < height; ++ y) {
    row_pointers[y] = reinterpret_cast<png_bytep>(image->row(y));
  }
  png_read_image(png_ptr, row_pointers);
"

will convert 16-bit RGB image into 8-bit RGB right?

The libpng api: void PNGAPI
png_read_image(png_structrp png_ptr, png_bytepp image) require png_bytepp( uchar pointer pointer).

Is there any easy way to change the above code so that it supports 16-bit RGB image as input?

puzzlepaint, 2023-01-04:
The PNG loading code is written such that it always tries to convert the file's format to the format of the `Image` object which the file is loaded into. So, if that works as intended, with a 3-channel 16-bit `Image` used as output, loading the images might already work.

However, to actually use 16-bit RGB images in the SLAM algorithm would require adjusting other places in the code. As a start, this definition has `Vec3u8` (a 3-channel 8-bit unsigned char vector) hard-coded as color type and would need to be adjusted to something like `Eigen::Matrix<u16, 3, 1>`:
https://github.com/ETH3D/badslam/blob/c66bdc841658cc04f9feae229c746b32f4284102/applications/badslam/src/badslam/main.cc#L508

And then any potential other places that might assume that the color is 8 bits per channel need to be adjusted. For example, the color storage in keyframes:
https://github.com/ETH3D/badslam/blob/c66bdc841658cc04f9feae229c746b32f4284102/applications/badslam/src/badslam/keyframe.h#L230

Starry-lei, 2023-01-06:
I adjusted a lot of places that assume the color image is 8-bit RGB along  the dirction of data flow, but I got stuck here:
https://github.com/ETH3D/badslam/blob/c66bdc841658cc04f9feae229c746b32f4284102/libvis/src/libvis/cuda/cuda_buffer.cu#L154

Once I extend the "template class CUDABuffer_<uchar4>;" to "template class CUDABuffer_<ushort4>;", an error 

""Invalid operands to binary expression ('float4' and 'float')"  "

occurs at : 
 https://github.com/ETH3D/badslam/blob/c66bdc841658cc04f9feae229c746b32f4284102/libvis/src/libvis/cuda/cuda_buffer.cu#L89 

How do you think I should adjust this?
Hmm, does that mean I should override operator "*"? Does it make sense for next steps?

puzzlepaint, 2023-01-09:
No, look here instead:
https://github.com/ETH3D/badslam/blob/c66bdc841658cc04f9feae229c746b32f4284102/libvis/src/libvis/cuda/cuda_buffer.cu#L124

This list contains an entry for `uchar4`:
https://github.com/ETH3D/badslam/blob/c66bdc841658cc04f9feae229c746b32f4284102/libvis/src/libvis/cuda/cuda_buffer.cu#L132

So it probably makes sense to add an entry for `ushort4` there as well. This should cause the function containing line 89 not to be compiled for that type.