ETH3D/badslam #95: Error on Ubuntu 22 and Nvidia 4090

Dav1dch, 2023-09-07:
Can badslam run on Ubuntu 22.04 and with Nvidia 4090 gpu? I have successfully compile it but fail on running it.

`time                            file:line     v| 
12:40:44.344              loguru.cpp:587   INFO| arguments: ../build_RelWithDebInfo/applications/badslam/badslam
12:40:44.344              loguru.cpp:590   INFO| Current dir: /home/david/Apps/badslam/build_RelWithDebInfo
12:40:44.344              loguru.cpp:592   INFO| stderr verbosity: 2
12:40:44.344              loguru.cpp:593   INFO| -----------------------------------
12:40:44.345                 main.cc:477   WARN| No auto-tuning file found -> using default parameters. GPU performance is thus probably slightly worse than it could be.
12:40:55.770 render_window_qt_opengl:63     ERR| OpenGL Error: GL_INVALID_VALUE (1281), description:
A numeric argument is out of range. The offending command is ignored and has no other side effect than to set the error flag.
12:40:55.775 render_window_qt_opengl:63     ERR| OpenGL Error: GL_INVALID_VALUE (1281), description:
A numeric argument is out of range. The offending command is ignored and has no other side effect than to set the error flag.
12:40:55.777      gui_main_window.cc:1603  INFO| Read dataset with 98 frames
12:40:55.778 render_window_qt_opengl:63     ERR| OpenGL Error: GL_INVALID_VALUE (1281), description:
A numeric argument is out of range. The offending command is ignored and has no other side effect than to set the error flag.
Stack trace:
66      0x557b7c40dea5 ../build_RelWithDebInfo/applications/badslam/badslam(+0x3fea5) [0x557b7c40dea5]
65      0x7fc8f3829e40 __libc_start_main + 128
64      0x7fc8f3829d90 /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7fc8f3829d90]
63      0x557b7c4050b6 ../build_RelWithDebInfo/applications/badslam/badslam(+0x370b6) [0x557b7c4050b6]
62      0x557b7c41074b ../build_RelWithDebInfo/applications/badslam/badslam(+0x4274b) [0x557b7c41074b]
61      0x557b7c45c301 ../build_RelWithDebInfo/applications/badslam/badslam(+0x8e301) [0x557b7c45c301]
60      0x7fc8f42c0cf4 QCoreApplication::exec() + 148
59      0x7fc8f42b875b QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 299
58      0x7fc8f43130b8 QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 104
57      0x7fc8ef11e3e3 g_main_context_iteration + 51
56      0x7fc8ef176258 /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab258) [0x7fc8ef176258]
55      0x7fc8ef120d3b g_main_context_dispatch + 619
54      0x7fc8ed0f8d6e /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x73d6e) [0x7fc8ed0f8d6e]
53      0x7fc8f4716a2c QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 172
52      0x7fc8f473dbc0 QGuiApplicationPrivate::processGeometryChangeEvent(QWindowSystemInterfacePrivate::GeometryChangeEvent*) + 384
51      0x7fc8f42b9e3a QCoreApplication::notifyInternal2(QObject*, QEvent*) + 314
50      0x7fc8f4f6c713 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 131
49      0x7fc8f4fcc329 /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cc329) [0x7fc8f4fcc329]
48      0x7fc8f4fc83fe /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c83fe) [0x7fc8f4fc83fe]
47      0x7fc8f42b9e3a QCoreApplication::notifyInternal2(QObject*, QEvent*) + 314
46      0x7fc8f4f6c6f7 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 103
45      0x7fc8f4f8c2e8 QLayoutPrivate::doResize() + 296
44      0x7fc8f50d59d6 /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2d59d6) [0x7fc8f50d59d6]
43      0x7fc8f50d0075 /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2d0075) [0x7fc8f50d0075]
42      0x7fc8f509fd20 /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x29fd20) [0x7fc8f509fd20]
41      0x7fc8f50619f4 /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2619f4) [0x7fc8f50619f4]
40      0x7fc8f40b3c1c QAbstractAnimation::start(QAbstractAnimation::DeletionPolicy) + 396
39      0x7fc8f40b99b6 QPropertyAnimation::updateState(QAbstractAnimation::State, QAbstractAnimation::State) + 662
38      0x7fc8f40b59e8 /lib/x86_64-linux-gnu/libQt5Core.so.5(+0xb59e8) [0x7fc8f40b59e8]
37      0x7fc8f40b8d91 QPropertyAnimation::updateCurrentValue(QVariant const&) + 273
36      0x7fc8f4fb13c3 QWidget::qt_metacall(QMetaObject::Call, int, void**) + 99
35      0x7fc8f4fa6f30 QWidget::setGeometry(QRect const&) + 128
34      0x7fc8f4fa6031 QWidgetPrivate::setGeometry_sys(int, int, int, int, bool) + 689
33      0x7fc8f42b9e3a QCoreApplication::notifyInternal2(QObject*, QEvent*) + 314
32      0x7fc8f4f6c6f7 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 103
31      0x7fc8f4f8c2e8 QLayoutPrivate::doResize() + 296
30      0x7fc8f4f85a07 QBoxLayout::setGeometry(QRect const&) + 983
29      0x7fc8f4f9008a QWidgetItem::setGeometry(QRect const&) + 474
28      0x7fc8f4fa6f30 QWidget::setGeometry(QRect const&) + 128
27      0x7fc8f4fa6031 QWidgetPrivate::setGeometry_sys(int, int, int, int, bool) + 689
26      0x7fc8f42b9e3a QCoreApplication::notifyInternal2(QObject*, QEvent*) + 314
25      0x7fc8f4f6c6f7 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 103
24      0x7fc8f4f8c2e8 QLayoutPrivate::doResize() + 296
23      0x7fc8f50d59d6 /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2d59d6) [0x7fc8f50d59d6]
22      0x7fc8f50d0075 /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2d0075) [0x7fc8f50d0075]
21      0x7fc8f509fd20 /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x29fd20) [0x7fc8f509fd20]
20      0x7fc8f50619f4 /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2619f4) [0x7fc8f50619f4]
19      0x7fc8f40b3c1c QAbstractAnimation::start(QAbstractAnimation::DeletionPolicy) + 396
18      0x7fc8f40b99b6 QPropertyAnimation::updateState(QAbstractAnimation::State, QAbstractAnimation::State) + 662
17      0x7fc8f40b59e8 /lib/x86_64-linux-gnu/libQt5Core.so.5(+0xb59e8) [0x7fc8f40b59e8]
16      0x7fc8f40b8d91 QPropertyAnimation::updateCurrentValue(QVariant const&) + 273
15      0x7fc8f4fd0249 QOpenGLWidget::qt_metacall(QMetaObject::Call, int, void**) + 25
14      0x7fc8f4fb13c3 QWidget::qt_metacall(QMetaObject::Call, int, void**) + 99
13      0x7fc8f4fa6f30 QWidget::setGeometry(QRect const&) + 128
12      0x7fc8f4fa6031 QWidgetPrivate::setGeometry_sys(int, int, int, int, bool) + 689
11      0x7fc8f42b9e3a QCoreApplication::notifyInternal2(QObject*, QEvent*) + 314
10      0x7fc8f4f6c713 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 131
9       0x7fc8f4faff6c QWidget::event(QEvent*) + 3212
8       0x7fc8f4fcec2f QOpenGLWidget::resizeEvent(QResizeEvent*) + 175
7       0x7fc8f4fa743a QWidgetPrivate::sendPaintEvent(QRegion const&) + 58
6       0x7fc8f42b9e3a QCoreApplication::notifyInternal2(QObject*, QEvent*) + 314
5       0x7fc8f4f6c713 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 131
4       0x7fc8f4faf4ee QWidget::event(QEvent*) + 526
3       0x7fc8f4fcedcd /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cedcd) [0x7fc8f4fcedcd]
2       0x557b7c59fc43 ../build_RelWithDebInfo/applications/badslam/badslam(+0x1d1c43) [0x557b7c59fc43]
1       0x557b7c59e134 ../build_RelWithDebInfo/applications/badslam/badslam(+0x1d0134) [0x557b7c59e134]
0       0x7fc8f59f04aa loguru::StreamLogger::~StreamLogger() + 138
12:40:55.908        render_window.cc:972   FATL| Cuda Error: unknown error
[1]    3699608 IOT instruction (core dumped)  ../build_RelWithDebInfo/applications/badslam/badslam
`

mszuyx, 2023-11-15:
I have encountered a similar error here.

I am running with Ubuntu 20.04, 3070 GPU, driver 470, CUDA 10.1.

I also included the output from GDB as the original error message was not very helpful.

It seems like a CUDA / GPU driver issue. I have tried reinstalling them but no luck. Any help would be appreciated.

(gdb) run
Starting program: /home/yu.chen/badslam/build_RelWithDebInfo/applications/badslam/badslam 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffee75e700 (LWP 10446)]
time                            file:line     v| 
16:12:38.891              loguru.cpp:587   INFO| arguments: /home/yu.chen/badslam/build_RelWithDebInfo/applications/badslam/badslam
16:12:38.891              loguru.cpp:590   INFO| Current dir: /home/yu.chen/badslam
16:12:38.891              loguru.cpp:592   INFO| stderr verbosity: 2
16:12:38.891              loguru.cpp:593   INFO| -----------------------------------
[New Thread 0x7fffed01b700 (LWP 10447)]
[New Thread 0x7fffec81a700 (LWP 10448)]
[New Thread 0x7fffe5ac9700 (LWP 10450)]
[New Thread 0x7fffe4d4a700 (LWP 10451)]
[Thread 0x7fffe4d4a700 (LWP 10451) exited]
[New Thread 0x7fffe4d4a700 (LWP 10455)]
[New Thread 0x7fffda849700 (LWP 10456)]
[New Thread 0x7fffda048700 (LWP 10457)]
16:12:49.507      gui_main_window.cc:1603  INFO| Read dataset with 3931 frames
[New Thread 0x7fffd94c7700 (LWP 10460)]
[New Thread 0x7fffd8cc6700 (LWP 10461)]
[New Thread 0x7fffca8d7700 (LWP 10462)]
Stack trace:
31      0x55555558877e /home/yu.chen/badslam/build_RelWithDebInfo/applications/badslam/badslam(+0x3477e) [0x55555558877e]
30      0x7ffff3e2f083 __libc_start_main + 243
29      0x555555581091 /home/yu.chen/badslam/build_RelWithDebInfo/applications/badslam/badslam(+0x2d091) [0x555555581091]
28      0x55555558a354 /home/yu.chen/badslam/build_RelWithDebInfo/applications/badslam/badslam(+0x36354) [0x55555558a354]
27      0x5555555dd634 /home/yu.chen/badslam/build_RelWithDebInfo/applications/badslam/badslam(+0x89634) [0x5555555dd634]
26      0x7ffff503bd6b QWidgetPrivate::setVisible(bool) + 459
25      0x7ffff5038bf3 QWidgetPrivate::show_helper() + 83
24      0x7ffff5038b61 QWidgetPrivate::showChildren(bool) + 337
23      0x7ffff503bd6b QWidgetPrivate::setVisible(bool) + 459
22      0x7ffff5038bf3 QWidgetPrivate::show_helper() + 83
21      0x7ffff5038b61 QWidgetPrivate::showChildren(bool) + 337
20      0x7ffff503bd6b QWidgetPrivate::setVisible(bool) + 459
19      0x7ffff5038bf3 QWidgetPrivate::show_helper() + 83
18      0x7ffff5038b61 QWidgetPrivate::showChildren(bool) + 337
17      0x7ffff503bd6b QWidgetPrivate::setVisible(bool) + 459
16      0x7ffff5038bd7 QWidgetPrivate::show_helper() + 55
15      0x7ffff5034e6e QWidgetPrivate::sendPendingMoveAndResizeEvents(bool, bool) + 334
14      0x7ffff45e180a QCoreApplication::notifyInternal2(QObject*, QEvent*) + 394
13      0x7ffff50030f0 QApplication::notify(QObject*, QEvent*) + 816
12      0x7ffff4ff9a66 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 134
11      0x7ffff503c947 QWidget::event(QEvent*) + 2327
10      0x7ffff505ec2f QOpenGLWidget::resizeEvent(QResizeEvent*) + 175
9       0x7ffff5034f4a QWidgetPrivate::sendPaintEvent(QRegion const&) + 58
8       0x7ffff45e180a QCoreApplication::notifyInternal2(QObject*, QEvent*) + 394
7       0x7ffff50030f0 QApplication::notify(QObject*, QEvent*) + 816
6       0x7ffff4ff9a66 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 134
5       0x7ffff503c2b6 QWidget::event(QEvent*) + 646
4       0x7ffff505ee05 /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cfe05) [0x7ffff505ee05]
3       0x7ffff5610a72 vis::RenderWidgetOpenGL::paintGL() + 834
2       0x55555572596d /home/yu.chen/badslam/build_RelWithDebInfo/applications/badslam/badslam(+0x1d196d) [0x55555572596d]
1       0x5555557217f5 /home/yu.chen/badslam/build_RelWithDebInfo/applications/badslam/badslam(+0x1cd7f5) [0x5555557217f5]
0       0x7ffff5566b92 loguru::StreamLogger::~StreamLogger() + 162
16:12:49.744        render_window.cc:972   FATL| Cuda Error: unknown error

Thread 1 "badslam" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
[... 20 lines cut ...]
#12 0x00007ffff50030f0 in QApplication::notify(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x00007ffff45e180a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff5034f4a in QWidgetPrivate::sendPaintEvent(QRegion const&) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007ffff505ec2f in QOpenGLWidget::resizeEvent(QResizeEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff503c947 in QWidget::event(QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007ffff4ff9a66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff50030f0 in QApplication::notify(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#19 0x00007ffff45e180a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007ffff5034e6e in QWidgetPrivate::sendPendingMoveAndResizeEvents(bool, bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
[... cut ...] nt_frame_cloud=false, follow_camera=false, show_input_images=false, window_width=1280, window_height=720)
    at /home/yu.chen/badslam/applications/badslam/src/badslam/gui_main_window.cc:113
#33 0x000055555558a354 in __libvis_main(int, char**) (argc=<optimized out>, argv=0x7fffffffde98) at /home/yu.chen/badslam/applications/badslam/src/badslam/main.cc:522
#34 0x0000555555581091 in main(int, char**) (argc=1, argv=0x7fffffffde98) at /home/yu.chen/badslam/applications/badslam/src/badslam/main.cc:69

mszuyx, 2023-11-15:
Ok I tried upgrade / install CUDA 11.4 to match my driver and hardware. I also rebuilt OpenCV and DLib with the new CUDA then tried re-built bad slam with `sm_86`. Still no luck, same error.

mszuyx, 2023-11-16:
Today I tried building all dependencies again from the very beginning with OpenCV 4.5 and CUDA 11.4. Still the same error. I then tried using the docker image. If I build docker with architecture `sm_80+` it will cause: 
```
0.793 -- Check for working CUDA compiler: /usr/local/cuda/bin/nvcc -- broken
0.793 CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCUDACompiler.cmake:46 (message):
0.793   The CUDA compiler
0.793 
0.793     "/usr/local/cuda/bin/nvcc"
0.793 
0.793   is not able to compile a simple test program.
0.793 
0.793   It fails with the following output:
0.793 
0.793     Change Dir: /home/badslam/build_RelWithDebInfo/CMakeFiles/CMakeTmp
0.793     
0.793     Run Build Command(s):/usr/bin/make cmTC_1d3ea/fast && /usr/bin/make -f CMakeFiles/cmTC_1d3ea.dir/build.make CMakeFiles/cmTC_1d3ea.dir/build
0.793     make[1]: Entering directory '/home/badslam/build_RelWithDebInfo/CMakeFiles/CMakeTmp'
0.793     Building CUDA object CMakeFiles/cmTC_1d3ea.dir/main.cu.o
0.793     /usr/local/cuda/bin/nvcc    -arch=sm_87    -x cu -c /home/badslam/build_RelWithDebInfo/CMakeFiles/CMakeTmp/main.cu -o CMakeFiles/cmTC_1d3ea.dir/main.cu.o
0.793     nvcc fatal   : Value 'sm_87' is not defined for option 'gpu-architecture'
0.793     make[1]: *** [CMakeFiles/cmTC_1d3ea.dir/build.make:66: CMakeFiles/cmTC_1d3ea.dir/main.cu.o] Error 1
0.793     make[1]: Leaving directory '/home/badslam/build_RelWithDebInfo/CMakeFiles/CMakeTmp'
0.793     make: *** [Makefile:121: cmTC_1d3ea/fast] Error 2
0.793     
0.793     
0.793 
0.793   
0.793 
0.793   CMake will not be able to correctly generate this project.
0.793 Call Stack (most recent call first):
0.793   CMakeLists.txt:45 (enable_language)
0.793 
0.793 
0.793 -- Configuring incomplete, errors occurred!
0.793 See also "/home/badslam/build_RelWithDebInfo/CMakeFiles/CMakeOutput.log".
0.793 See also "/home/badslam/build_RelWithDebInfo/CMakeFiles/CMakeError.log".
------
Dockerfile:101
--------------------
  99 |     ARG CUDA_ARCH
 100 |     ENV CUDA_ARCH ${CUDA_ARCH}
 101 | >>> RUN cmake -E env CXXFLAGS='-march=native' cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_CUDA_FLAGS="-arch=${CUDA_ARCH}" .. && make -j
 102 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c cmake -E env CXXFLAGS='-march=native' cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_CUDA_FLAGS=\"-arch=${CUDA_ARCH}\" .. && make -j" did not complete successfully: exit code: 1
```
If I do `sm_75` (even though I think 3070 belongs to `Ampere`) it will take me further and eventually fail at:
```
40.43 In file included from /usr/local/include/g2o/core/base_binary_edge.h:30,
40.43                  from /usr/local/include/g2o/types/slam3d/edge_se3.h:30,
40.43                  from /home/badslam/applications/badslam/src/badslam/pose_graph_optimizer.cc:36:
40.43 /usr/local/include/g2o/core/base_fixed_sized_edge.h:38:10: fatal error: ceres/internal/fixed_array.h: No such file or directory
40.43    38 | #include <ceres/internal/fixed_array.h>
40.43       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
40.43 compilation terminated.
40.43 make[2]: *** [applications/badslam/CMakeFiles/badslam_baselib.dir/build.make:655: applications/badslam/CMakeFiles/badslam_baselib.dir/src/badslam/pose_graph_optimizer.cc.o] Error 1
40.43 make[2]: *** Waiting for unfinished jobs....
```

I then tried using the Windows executable and it works perfectly with saved dataset and Realsense live input.

However, getting Bad SLAM working on Ubuntu is the whole point for us. @puzzlepaint Dear friend, do you have any idea how to fix the runtime error / docker build error?

antmusco, 2024-02-14:
I was having the same issue, it was resolved on my end by installing `vainfo` and `mesa-va-drivers` to allow an Nvidia-based OpenGL context. Opened the PR above to fix it.